cta-observatory/ctapipe #1721: backward compatibility of data model

kosack, 2021-05-16:
When we merge #1426 for example, we will break reading data files produced with a previous version of ctapipe. 

This brings up the question: *should we support some level of backward compatibility in DL1/DL2 outputs and how?*

This has already been an issue when we have updated the *instrument* data model in the past, and something we should be careful about.  Since we store both the table version (TAB_VER) for instrument outputs, and the DATA MODEL VERSION for the DL1/DL2 files themselves, we should be able to support reading old data, at the expense of keeping around some old code (e.g. renaming `from_table()`  to `from_table_v1()` and having `from_table()` dispatch the correct version.

Clearly we want this after v1.0.0 of ctapipe, but since people are using DL1 files now, it could already be critical.

maxnoe, 2021-05-16:
@kosack No, you can still read them. I build it in a backwards compatible way.

bregeon, 2021-05-17:
just a note that it's fast enough to reprocess "simulated" reference datasets with a new version of ctapipe at any time, so loosing backwards compatibility at some point should not be an issue from this point of view... however it might be a different story for LST1 data.

kosack, 2021-05-17:
I think some backward compatibility is necessary, as it's a bad user experience if we cannot repeat a particular analysis because the data model changed (otherwise it can be hard to compare benchmarks between versions)

maxnoe, 2023-01-04:
We now version the data model and raise clear errors for unsupported versions. We still might to need breaking changes in future, so I think this is all we can do for now.