cta-observatory/ctapipe #1729: [FR - ctapipe-merge]: Add option 'split_datasets_by'

TjarkMiener, 2021-05-27:
Would be great to merge tel_types at this stage of the pipeline.

LukasNickel, 2021-05-27:
I am not sure I understand what you are trying to achieve.
You want to merge dl1 files, but only specific telescope types? Or do you want to group telescope types in the output?

TjarkMiener, 2021-05-27:
I want to group the telescope types in the output with stage1 h5 files as input. The output file looks like this (hopefully the same as if you would run stage1 with `--split_datasets_by tel_type` on simtel files):
[image]

LukasNickel, 2021-05-27:
I am not sure if this is intended/wanted.
Maybe @kosack can comment.

Why do you need this? It looks organized if you want to work on one telescope type at a time of course, but you should always be able to combine the tables in your own code, especially as this does not even seems to be ordered by trigger time

maxnoe, 2021-05-27:
@LukasNickel this option is supported by the stage1 tool currently. I think it makes sense to also have it for the merge tool.

We can discuss if we want to keep this option at all, but having this option only for some tools and not all is not great.

vuillaut, 2021-05-27:
> @LukasNickel this option is supported by the stage1 tool currently. I think it makes sense to also have it for the merge tool.
> 
> We can discuss if we want to keep this option at all, but having this option only for some tools and not all is not great.

Does that mean that there are several DL1 data models?

LukasNickel, 2021-05-27:
Oh it is supported? In that case we can add it here of course.

maxnoe, 2021-05-27:
That's splitting hairs :D 

You could call it one data model with two allowed structures...

But you are right. And I wouldn't recommend using the "tel_type" option. See e.g. this connected issue #971, #1426 and some more related to this.

vuillaut, 2021-05-27:
> > Does that mean that there are several DL1 data models?
> 
> That's splitting hairs :D
> 
> You could call it one data model with two allowed structures...
> 
> But you are right. And I wouldn't recommend using the "tel_type" option. See e.g. this connected issue #971, #1426 and some more related to this.

Is there a big advantage (reading speed?) having telescopes merged per tel_type?
Having a method `read_dl1_table_per_tel_type` or something similar should be enough to cover the "ease of use" advantage.

LukasNickel, 2021-05-27:
Is this splitting code tested anywhere? I feel like this might break at least the dl1eventsource

maxnoe, 2021-05-27:
@LukasNickel If the DL1 event source does not support reading those files, I would be in favor of just removing that option, due to the other mentioned reasons.

TjarkMiener, 2021-05-27:
It might effect our (DL) reading speed. I'm currently testing this.

The `read_table` function should work out of the box, no?

maxnoe, 2021-05-27:
@TjarkMiener yes, read table will work on everything written with the data writer.

I don't expect a large difference between:

```
images = vstack([
    read_table(path, f'/dl1/event/telescope/images/tel_{tel_id:03d}')
    for tel_id in subarray.tel
])
```
and just reading the single table, but I have not measured it.
    
```

LukasNickel, 2021-05-27:
> @LukasNickel If the DL1 event source does not support reading those files

I would need to test, but since I have never seen this option before I have certainly not taken it into account and I dont see any tests at first glance

kosack, 2021-05-27:
> Is there a big advantage (reading speed?) having telescopes merged per tel_type?
Having a method read_dl1_table_per_tel_type or something similar should be enough to cover the "ease of use" advantage.

It was mainly there for ease of certain use cases (e.g. training of any ML technique on a per-telescope basis), but it is true that using `vstack` to merge a set of tables is not much slower, and only request a few more lines of code at  read time, so we could consider just removing the group_by option.  However, the other reason I had it there originally was for very large super-arrays, where there may be cases where there are performance issues when writing so many tables (we already run into some limits of pytables tthat needed to be increased).  Also, when using `ctapipe-merge` or `DL1EventSource` these files are very slow they have to construct 500+ table schemas.  If this is just for training from simulations, it can be nice to have it split by tel type just to reduce the number of datasets (and the metadata overhead).   For observed data, we never would group by telescope type of course.

HealthyPear, 2021-06-03:
In the reference analysis of simulated data we always to the ML training by telescope types, so:
- either we keep this option (I always linked the data model to the variable naming rather than the tables structure in the files, but I understand the latter can be important also)
- provide a `read_dl1_table_per_tel_type` function in ctapipe so any software / benchmark uses that instead of re-inventing the wheel each time

maxnoe, 2022-07-04:
We discussed this again in the dev meeting in Annecy and decided we want to stop supporting the `by telescope type` structure and only support the single table for each telescope moving forward.

closed 2022-07-04 by maxnoe