cta-observatory/ctapipe #1738: Stage1 tool's `--write-parameters` flag is superflous

maxnoe, 2021-06-01:
The stage1 tool has a flag `-write-paramters` that's not useful since it can only set `DataWriter.write_parameters` to True, which already is the default.

Flags are only useful if they set values other than the default.

There is also currently no nice way to deactivate writing parameters, so the flag should probably be `--no-parameters` instead.

We could also follow a common pattern and provide pairs of flags for everything, so I would propose to change

`--write-images`, `--write-paramters` to `--images`, `--no-images`, `--parameters`, `--no-parameters`.

kosack, 2021-06-01:
I'm actually surprised that traitlets.config's "boolean flag" functionality doesn't automatically add negative flags... it really should (e.g. `--no-write-images`).  Or we make the defaults all False maybe?

kosack, 2021-06-01:
In any case, we may have to think about what these really mean in the future: I called them `write-X` for a reason: they do not tell the tool if it should *compute* them (that depends on  the input), only if they should be in the output.

nbiederbeck, 2021-06-02:
> I'm actually surprised that traitlets.config's "boolean flag" functionality doesn't automatically add negative flags... it really should (e.g. `--no-write-images`). Or we make the defaults all False maybe?

Actually it does, when using `traitlets.config.boolean_flag` (or `ctapipe.core.traits.flag` as we call it):
```
flag("write-parameters", "DataWriter.write_parameters", "store parameters in output", "dont store"))
```
yields the following `--help` output:
```
--write-parameters
    store parameters in output
    Equivalent to: [--DataWriter.write_parameters=True]
--no-write-parameters
    dont store
    Equivalent to: [--DataWriter.write_parameters=False]
```

maxnoe, 2021-06-02:
Ha, nice. Can you make a PR fixing that for our tools?

nbiederbeck, 2021-06-02:
Will do

PR #1748 (2021-06-04): Add no-flags, add short aliases, make aliases more consistent
Fixes #1738
closed 2021-06-04 by maxnoe