cta-observatory/ctapipe #1743: Completely remove 2d geometries

maxnoe, 2021-06-03:
I think it will make things much more consistent, if we stick with 1d geometries, as all the normal geometries are this way and it will make implementing more geometry features easier if we only have one dimensioniality to worry about.

To keep the geometry conversion functionality, I would propose to make sure that this results in a row major array, so deep learning and other algorithms requiring classical images can just do

``` 
image.reshape(converted_geometry.shape2d)
```

or 
```
image2d = geometry.to_2d_image(image)
```

kosack, 2021-06-03:
sounds ok, then the GeometryConverter that makes "skewed" geometries now would just return a new 1D array with he shape. I think that is ok.

maxnoe, 2023-01-04:
Done in #1728

closed 2023-01-04 by maxnoe