cta-observatory/ctapipe #1755: Machine Learning Module

maxnoe, 2021-07-06:
At least for particle classification, energy regression and for the mono case also for direction reconstruction, we need to implement Components / Tools to train and apply machine learning models.

This is currently not implemented in ctapipe, but in several tools including `protopipe.mva`, `aict-tools`, `lstchain.reco` and probably more.

We should implement this in ctapipe, combining what we learned from those above, using the ctapipe configuration system.

A first step should probably focus on sklearn models on image parameters, a second step could implement an abstract API for training and applying deeplearning models, for which we could use a plugin system similar to IO so that for example the deep learning people can try new architectures / frameworks inside ctapipe.

Needed targets:
- [ ] Energy Regression
- [ ] Particle Classification
- [ ] Origin using the disp method
- [ ] Combination of mono predictions

Needed Tools:
- Training

Components:
- Application of each of the models / combination of mono predictions (to be used inside `ShowerReconstructor` and/or using tables.)

Most of this can probably be taken directly from aict-tools / protopipe.mva adapted to Components the config system.

kosack, 2021-07-06:
To add to that list:

-  [ ] Event quality (type) classification and/or reconstruction merging

kosack, 2021-07-06:
this is related to #1744

maxnoe, 2023-02-03:
Since most of this is now implemented, I think it makes sense to close this large issue and open smaller ones for the remaining tasks.