cta-observatory/ctapipe #1759: `Path` trait does not allow to set `allow_none` attribute to `True`

mexanick, 2021-07-23:
From the source code it appears that it is not possible to set the attribute `allow_none` to` False` and the validation of empty path with attribute `exists` set to `True` doesn't make sense.
Reasons:
https://github.com/cta-observatory/ctapipe/blob/37d901581c478bdfcd56c699feba98588c371046/ctapipe/core/traits.py#L117
prevents setting the `allow_none` attribute:
TypeError: __init__() got multiple values for keyword argument 'allow_none'

and 
https://github.com/cta-observatory/ctapipe/blob/37d901581c478bdfcd56c699feba98588c371046/ctapipe/core/traits.py#L147-L151

will return always `None` if the default value is not set.

Is such behavior is expected? I'd change the hardcoded `allow_none` to a default value with a possibility to override...

_Originally posted by @mexanick in https://github.com/cta-observatory/ctapipe/issues/1758#issuecomment-885000278_

Original answer from @maxnoe

https://github.com/cta-observatory/ctapipe/pull/1758#issuecomment-885029874

maxnoe, 2021-07-23:
> Is such behavior is expected? I'd change the hardcoded allow_none to a default value with a possibility to override...

I couldn't make `allow_none=False` work, so at the moment, you have to allow None for `Path` traits and check later if it is None and raise an error if you do want to require it.

A workaround is to define a default like this:

```
from traitlets import default, TraitError
from ctapipe.core import Component
from ctapipe.core.traits import Path

class Foo(Component):
    path = Path().tag(config=True)

    @default("path")
    def path_default(self):
        raise TraitError("path is required")

f = Foo()
print(f.path)
```

maxnoe, 2021-07-23:
Ok, digging  deeper again into how traitlets handles defaults and validation, I think I came up with a solution so that Path can use `allow_none=False` and it produces meaningful errors.

See #1760

commit 72654e65 (2021-07-23): Implement allow_none=False for Path trait, fixes #1759
commit 69c2bc94 (2021-07-26): Implement allow_none=False for Path trait, fixes #1759 (#1760)

* Implement allow_none=False for Path trait, fixes #1759

* Initialize logger first in component to make it available in observe

* Make repr_html work after tool has been run
closed 2021-07-26 by kosack