cta-observatory/ctapipe #1761: Set default contact name to current user?

kosack, 2021-07-23:
We could add a default contact name to DataWriter in case the user doesn't specify one in a config file that just gives the current user's full name, e.g.:
```py3
import os
import pwd
default = pwd.getpwuid(os.getuid()).pw_gecos
```
That would make it less error-prone track the provenance of files created by users.  In productions, the config file should set it properly however. This info is already in the provenance.log, but not in the reference metadata in the file itself. 

not sure if that works on Windows, but we don't really support windows.

maxnoe, 2021-07-23:
> This module provides access to the Unix user account and password database. It is available on all Unix versions.

kosack, 2021-07-23:
I suppose we can just wrap it in a `try:catch`, and if it works, great, if not on windows or similar, it's not a big issue.   It's best to set a good default if possible, otherwise fall back to None. The idea is only to make it less error-prone if the  user forgets to fill in the info.

kosack, 2021-08-24:
this may be redundant if #1774 is merged - it forces the user to provide the contact info in the generated config files

kosack, 2021-08-25:
implemented in #1774

closed 2021-08-25 by kosack