cta-observatory/ctapipe #1768: Add reading functions to ctapipe.io

HealthyPear, 2021-07-29:
I propose to add a list of classes/methods or functions to read DL1 and DL2 files.

For example, the DL1 benchmarking workshop repo has a code snippet in `benchmark_helpers.py` called `DL1Loader` which just now I was about to copy-paste in `protopipe`, but there is no reason (in general) to copy-paste code from different sources or to reinvent the wheel everytime (especially since we decide our data formats).

I think we should provide here - under `ctapipe.io` - all necessary methods to read images and/or parameters recorded with both `tel_id` and `tel_type` file structures, get all unique existent camera names from the files, etc.

maxnoe, 2021-07-29:
@HealthyPear I will work on that for #1767, where this is needed.

maxnoe, 2021-07-29:
> get all unique existent camera names from the files, etc.

That's just `SubarryDescription.from_hdf(...)`

HealthyPear, 2021-07-29:
Though, since

- such feature is used /can be used) already for benchmarking purposes
- the timeline of #1767 is not clear
- these functions should be realtively easy to add (the tel_id versions are basically already in the workshop materials)

I would suggest to open a separate PR which I'm confident would be merged anyway before #1767

HealthyPear, 2021-07-29:
Perfect.

I am testing these functions with data produced from #1726 for review of that PR and to upgrade protopipe.mva in preparation of that PR so I can propose a first set of functions/classes.

HealthyPear, 2021-07-29:
It would be probably good to discuss here how we want the output to look like, since I realized there are different ways to do it.
For now I am experimenting with a single table for `tel_id`-based datasets and a `tel_type`-based dictionary of single tables for `tel_type`-based datasets.

By _single_ table I mean the reult of joining trigger, simshower, dl1 and dl2, etc...
Since in some cases it would be too much I am testing also a data-type mask to export only some of the information with one function/method.

another solution would be to have 1 function/method per-data-type (so 1 for trigger, 1 for dl2, etc...) and then the joining is done sequentially on what has been activated via the mask.

closed 2021-09-20 by kosack