cta-observatory/ctapipe #1778: Exception when array pointing is >90 deg due to floating point error

kosack, 2021-08-24:
`HillasReconstructor` throws an exception when converting the telescope pointing to an astropy.coordinate in the case where tthe pointing altitude is slightly greater than 90 deg, in this case 90.00000250447816 deg (which is probably a floating point error). 

To reproduce, use the file linked in #1777 and run:
```
ctapipe-process --config training_config.json \
    --input gamma_HESS_example.simhess.gz  \
    --output gamma_HESS_example.DL1DL2.h5
```

```
...
  File "/Users/kkosack/Projects/CTA/Working/ctapipe/ctapipe/reco/hillas_reconstructor.py", line 195, in __call__
    array_pointing = SkyCoord(
  File "/Users/kkosack/miniconda3/envs/cta-dev/lib/python3.8/site-packages/astropy/coordinates/sky_coordinate.py", line 314, in __init__
    skycoord_kwargs, components, info = _parse_coordinate_data(
  File "/Users/kkosack/miniconda3/envs/cta-dev/lib/python3.8/site-packages/astropy/coordinates/sky_coordinate_parsers.py", line 245, in _parse_coordinate_data
    valid_components.update(_get_representation_attrs(frame, units, kwargs))
  File "/Users/kkosack/miniconda3/envs/cta-dev/lib/python3.8/site-packages/astropy/coordinates/sky_coordinate_parsers.py", line 592, in _get_representation_attrs
    valid_kwargs[frame_attr_name] = repr_attr_class(value, unit=unit)
  File "/Users/kkosack/miniconda3/envs/cta-dev/lib/python3.8/site-packages/astropy/coordinates/angles.py", line 536, in __new__
    self._validate_angles()
  File "/Users/kkosack/miniconda3/envs/cta-dev/lib/python3.8/site-packages/astropy/coordinates/angles.py", line 557, in _validate_angles
    raise ValueError('Latitude angle(s) must be within -90 deg <= angle <= 90 deg, '
ValueError: Latitude angle(s) must be within -90 deg <= angle <= 90 deg, got 90.00000250447816 deg
2021-08-24 18:10:16,616 INFO [ctapipe.ctapipe-process] (tool.write_provenance): Output: /Users/kkosack/Projects/CTA/PipeWork/gamma_HESS_example.DL1DL2.h5
```

Note that some non-CTA telescopes (e.g. HESS CT5) can physically move above 90 degrees elevation (into reverse-pointing mode), which is another related issue, but here I think it is just enough to fix the case where numerical error causes it to be too high.

maxnoe, 2022-01-21:
I guess this should happen in the simtel event source already. We should validate / parse inputs at the IO boundaries, not somewhere in between.

maxnoe, 2022-09-21:
Same issue as https://github.com/cta-observatory/ctapipe/issues/2075

maxnoe, 2023-01-12:
Should have been closed automatically, but apparently `fixes #2075, #1778` does not close both issues.

closed 2023-01-12 by maxnoe