cta-observatory/ctapipe #1779: doc action failing

kosack, 2021-08-25:
Since a few weeks ago at least, the docs action in the CI is failing for all new PRs and the master, e.g.:
https://github.com/cta-observatory/ctapipe/actions/runs/1159019166

Seems to be some change in the sphinx-action perhaps.

HealthyPear, 2021-08-25:
also for reference: downgrading the action to the latest tagged release doesn't help - perhaps we should consider opening an issue there?

kosack, 2021-08-25:
I think the issue is our `pre-build` step, not the action itself.  The problem may be unrelated to the sphinx action, but to the docker container or something (update to ubuntu, etc).

kosack, 2021-08-30:
There is also the warning:
```
WARNING: apt does not have a stable CLI interface. Use with caution in scripts
```
Not sure what the alternative would be for installing things though! (and apt has been around for quite a while, so not sure why it would still be considered unstable... maybe they prefer `apt-get` directly, rather than the `apt` wrapper?)

maxnoe, 2021-08-30:
@kosack that warning is there since always and forever (and yes, I think it's the distinction between `apt` and `apt-get`).

The real issue here is the release of a new debian version, that somehow makes the `update` step fail and then the install fails because no package lists are available.

closed 2021-08-30 by kosack