cta-observatory/ctapipe #1785: Messed up metadata reading from multi-container tables

LukasNickel, 2021-09-09:
Reading containers from a table with multiple containers in it leads to duplicated metadata, because the `HDF5TableReader` adds all available metadata to all containers. This has two implications:
- The containers in memory have more metadata than they should have
- If a container is saved in multiple places, the output from running ctapipe-process on a hdf5 file (e.g. to get dl2 data from a dl1 file), can have messed up metadata. I found this because the `EventIndexContainer` had metadata from the `TriggerContainer`, which in turn was saved in other tables with the event index.
- `read_table` fails with such broken files, because it tries to perform transforms based on metadata entries, which dont have matching columns

This should reproduce it:
```python
from ctapipe.io import HDF5TableReader, HDF5TableWriter
from ctapipe.containers import HillasParametersContainer, EventIndexContainer

H = HillasParametersContainer(
    fov_lon=1 * u.deg, fov_lat=1 * u.deg, length=1 * u.deg, width=1 * u.deg
)
I = EventIndexContainer(obs_id=1, event_id=2)

with HDF5TableWriter("test.h5") as writer:
    writer.write("table", [H, I])

with HDF5TableReader("test.h5") as reader:
    _ = reader.read("/table", [HillasParametersContainer(), EventIndexContainer()])
    H2, I2, = next(_)

print(I2.meta)
```

Relevant piece in the `HDF5TableReader`:
https://github.com/cta-observatory/ctapipe/blob/63e0df4337c18386aab3dea0d0654c5ec02b447a/ctapipe/io/hdf5tableio.py#L489-L490

I guess there should be some kind of check to only read the relevant parts.

kosack, 2021-09-10:
Probably we should differentiate between "user-defined" metadata, and the "ctapipe column metadata" (e.g. `<colname>_DESC`).   

Right now there is not an easy way to get a list of all possible column metadata suffixes, but we could hard-code one, and then ignore any metadata with those.

nbiederbeck, 2022-06-24:
I also came across this bug when running multiple DL1 → DL1 conversion (for different cleaning levels with @aknierim).

I discussed a little bit with @LukasNickel offline and we think a solution where we check for a set of suffixes (e.g. `_DESC`, `_TRANSFORM` etc.) wether the correspondigng prefix exists in the container it should be written to should work. Example: if the table metadata has `time_TRANSFORM`, but the container has no field `time`, it should not be appended to the container metadata.

I would like to fix this problem asap, is there any more input you have for the solution?

maxnoe, 2022-07-04:
> Probably we should differentiate between "user-defined" metadata, and the "ctapipe column metadata" (e.g. <colname>_DESC)

Which would be easier if we'd use also a prefix, not just a suffix. Since we broke the data model for the current release already and plan to implement #1983, this would actually be a good time to do that.

HDF5 defines it actually using indices, not names: https://docs.hdfgroup.org/hdf5/v1_12/_t_b_l.html

```
In [11]: f.root.dl1.event.subarray.trigger.attrs
Out[11]: 
/dl1/event/subarray/trigger._v_attrs (AttributeSet), 26 attributes:
   [CLASS := 'TABLE',
    CTAPIPE_VERSION := '0.15.1.dev144+g29d23500',
    FIELD_0_FILL := 0,
    FIELD_0_NAME := 'obs_id',
    FIELD_1_FILL := 0,
    FIELD_1_NAME := 'event_id',
    FIELD_2_FILL := 0.0,
    FIELD_2_NAME := 'time',
    FIELD_3_FILL := False,
    FIELD_3_NAME := 'tels_with_trigger',
    FIELD_4_FILL := 0,
    FIELD_4_NAME := 'event_type',
    NROWS := 7,
    TITLE := 'Storage of EventIndexContainer,TriggerContainer',
    VERSION := '2.7',
    event_id_DESC := 'event identifier',
    event_type_DESC := 'Event type',
    event_type_ENUM := <enum 'EventType'>,
    event_type_TRANSFORM := 'enum',
    obs_id_DESC := 'observation identifier',
    tels_with_trigger_DESC := 'List of telescope ids that triggered the array event',
    tels_with_trigger_TRANSFORM := 'tel_list_to_mask',
    time_DESC := 'central average time stamp',
    time_TIME_FORMAT := 'mjd',
    time_TIME_SCALE := 'tai',
    time_TRANSFORM := 'time']
```

I would not use the same prefix, as that could collide with future extensions to this table standard, but it might work better than our current approach to use:

```
CTAFIELD_0_UNIT
CTAFIELD_0_TIME_FORMAT
```

etc.

closed 2022-07-19 by maxnoe