cta-observatory/ctapipe #1788: Unit tests fail because my name has non ascii characters

maxnoe, 2021-09-17:
The unit tests currently fail on my system (`ctapipe.tools.tests.test_tools::test_ctapipefileinfo`) because I have an umlaut in my name and writing the file info to FITS fails with an could not encode to ascii error.

Seriously....?

maxnoe, 2021-09-17:
Somehow it gets my name magically from the system.

The error was not even printed.

This took me an hour to find out that MY NAME is the problem

maxnoe, 2021-09-17:
And fileinfo depends on pandas although we removed pandas from our dependencies...

closed 2021-09-17 by kosack

kosack, 2021-09-20:
It's always good to work with non-English names when unicode/ascii is involved :-) 

See also: https://www.kalzumeus.com/2010/06/17/falsehoods-programmers-believe-about-names/