cta-observatory/ctapipe #1797: TableLoader cannot load flatfields

LukasNickel, 2021-09-30:
All flatfield events have event id -1, which makes them not uniquely identifiable by the keys `["obs_id", "event_id"]`.
The joins in the `TableLoader` will then lead to astropy generating a table with all possible combinations of rows (Merging two tables with 1000 rows each results in a table with 1000*1000 rows).
I don't see a straightforward solution besides giving them unique ids as well. You might be able to `hstack` the tables (probably the order of operations matters here to guarantee the tables have the same number of rows), but i am not confident that this wouldnt break at some point

Minimal example:
```python
from astropy.table import Table, join
t1 = Table({"merge_key": [1, 1, 1], "data": [1, 2, 3]})
t2 = Table({"merge_key": [1, 1, 1], "data2": [4, 5, 6]})
join(t1, t2, keys=["merge_key"])

<Table length=9>
merge_key  data data2
  int64   int64 int64
--------- ----- -----
        1     1     4
        1     1     5
        1     1     6
        1     2     4
        1     2     5
        1     2     6
        1     3     4
        1     3     5
        1     3     6

```

or load two tables from a file generated with
```
"SimTelEventSource":{"skip_calibration_events": false}
```
in the config

maxnoe, 2021-09-30:
Maybe a solution would be to give them event_ids in the SimTelEventSource? Using negative event ids for calibration events should work.

maxnoe, 2023-01-03:
This should be fixed, we gave calib events negative event ids in: https://github.com/cta-observatory/ctapipe/pull/1929

closed 2023-01-03 by maxnoe