cta-observatory/ctapipe #1805: matplotlib 3.5 breaks some code

kosack, 2021-11-16:
We can either: 
1. for now fix the version to 3.4
2. update the code (preferred)

Main issue seems to be a  `ValueError: Invalid vmin or vmax` when changing a CameraDisplay to log scale.

HealthyPear, 2021-11-16:
If it's a small thing like this I also vote for just fix it and support the newer release

kosack, 2021-11-16:
I can't seem to  reproduce it... 

```
> ctapipe-info --dependencies

*** ctapipe core dependencies ***

             astropy -- 4.3.1
             eventio -- 1.5.1.post1
             iminuit -- 2.7.0
          matplotlib -- 3.5.0
               numba -- 0.54.1
               numpy -- 1.20.3
              pytest -- 6.2.4
               scipy -- 1.7.1
             sklearn -- 1.0.1
              tables -- 3.6.1
           traitlets -- 5.1.1

*** ctapipe optional dependencies ***

   ctapipe_resources -- not installed
            graphviz -- not installed
          matplotlib -- 3.5.0
              pytest -- 6.2.4
```

```
> pytest -q ctapipe/visualization/
======================================================= test session starts ========================================================
platform darwin -- Python 3.8.12, pytest-6.2.4, py-1.10.0, pluggy-0.13.1
rootdir: /Users/kkosack/Projects/CTA/Working/ctapipe, configfile: setup.cfg
plugins: cov-2.12.1
collected 17 items

ctapipe/visualization/tests/test_bokeh.py .........                                                                          [ 52%]
ctapipe/visualization/tests/test_mpl.py ........                                                                             [100%]

======================================================== 17 passed in 7.44s ========================================================
```

maxnoe, 2021-11-16:
I can and I have the fix

kosack, 2021-11-16:
Ok great - maybe it's an interaction with some other dependency? Somehow it works fine for me.

commit 4206b805 (2021-11-16): Merge pull request #1807 from cta-observatory/mpl35

 Fix negative vmin/vmax raising error with mpl3.5, fixes #1805
commit b0a1bc68 (2021-11-16): Fix negative vmin/vmax raising error with mpl3.5, fixes #1805
closed 2021-11-16 by maxnoe