cta-observatory/ctapipe #1812: CameraGeometry position_to_pix_index exceptions handling

mexanick, 2021-12-01:
`position_to_pix_index` method handles identically cases where the position is outside of the camera and cases where position is in the gap between pixels. In both cases warning `Coordinate (x,y) lies outside camera` is raised.

Can this be changed to differentiate the outlined cases?

maxnoe, 2022-01-21:
Maybe just a better error message? `No pixel at position ...`? Or do you really need to be able to distinguish these two cases? 

It's not that easy for the irregular camera geometries we have

kosack, 2022-01-21:
I guess if it's really required, we could construct a convex hull using the vertices of the boundary pixels of each camera and test if we are inside or outside, but I agree with @maxnoe that maybe it's not really necessary? Or is there a strong use case for knowing if you are in a gap vs outside the camera?

maxnoe, 2022-01-24:
This could use shapely as demonstrated here: 

https://github.com/cta-observatory/ctapipe/issues/855#issuecomment-496578056

kosack, 2022-01-24:
You could use scipy.spatial instead to avoid a new dependency on shapely. Requires a bit more code, but doesn't seem too hard.  I found this example on StackOverflow:

https://stackoverflow.com/questions/16750618/whats-an-efficient-way-to-find-if-a-point-lies-in-the-convex-hull-of-a-point-cl

maxnoe, 2022-01-24:
But we don't want the *convex* hull, right? Just *the* hull. Which requires knowledge about the connections, not just the points.

maxnoe, 2022-01-24:
I.e. the convex hull for chec would include a diagonal line from the outer edges of the modules like this:

[image]

mexanick, 2022-01-24:
For my particular use case a convex hull based differentiation would be enough. And differentiating between "in (convex) hull" vs "out of (convex) hull" is desirable, but not too much important for me. As of now I simply ignore the raised warnings, however it is a sort of unsafe practice, as I know for a fact that in my case the normal behavior is still to have a point inside the hull.