cta-observatory/ctapipe #1813: Use of `/` inconsistent in io

maxnoe, 2021-12-03:
On the one side:

```
Traceback (most recent call last):
  File "/home/maxnoe/.local/anaconda/envs/lst-dev/lib/python3.8/site-packages/ctapipe/core/tool.py", line 289, in run
    self.finish()
  File "/home/maxnoe/Uni/CTA/cta-lstchain/lstchain/tools/lstchain_create_drs4_pedestal_file.py", line 259, in finish
    writer.write(key, drs4_calibration)
  File "/home/maxnoe/.local/anaconda/envs/lst-dev/lib/python3.8/site-packages/ctapipe/io/hdf5tableio.py", line 330, in write
    self._setup_new_table(table_name, containers)
  File "/home/maxnoe/.local/anaconda/envs/lst-dev/lib/python3.8/site-packages/ctapipe/io/hdf5tableio.py", line 257, in _setup_new_table
    raise ValueError("Table name must not start with '/'")
ValueError: Table name must not start with '/'
```

On the other side:

```
t = read_table('/tmp/dl1.h5', 'dl1/event/telescope/paramters/tel_001')
---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
<ipython-input-2-b15536436921> in <module>
----> 1 t = read_table('/tmp/drs4.h5', 'r1/')

~/Uni/CTA/ctapipe/ctapipe/io/astropy_helpers.py in read_table(h5file, path, start, stop, step, condition)
     69             )
     70 
---> 71         table = h5file.get_node(path)
     72         transforms, descriptions, meta = _parse_hdf5_attrs(table)
     73 

~/.local/anaconda/envs/cta-dev/lib/python3.7/site-packages/tables/file.py in get_node(self, where, name, classname)
   1636         elif isinstance(where, (str, numpy.str_)):
   1637             if not where.startswith('/'):
-> 1638                 raise NameError("``where`` must start with a slash ('/')")
   1639 
   1640             basepath = where

NameError: ``where`` must start with a slash ('/')
```

kosack, 2021-12-06:
The main reason was that `TableWriter` can create a hierarchy that is anywhere in an HDF file, not necessarily at the root, so the table names are *relative* to the base node (which is root by default.     Whereas if you want to request a specific node to read in an existing HDF5 file, you need to specify it from the root node, not as a relative path.  

So that is the main difference, and maybe should be documented better.  Perhaps renaming the arguments to relative_path in TableWriter?

maxnoe, 2021-12-06:
What is the harm in removing the slash if it is given in the table writer?

vuillaut, 2021-12-06:
Is there an issue with having the reader always add `/` as prefix?
`tables` or `astropy` do not require it, it would be consistent with users  habits

maxnoe, 2021-12-06:
Also fine

maxnoe, 2021-12-06:
> tables or astropy do not require it, it would be consistent with users habits

Actually the error comes from `tables`, `tables.File.get_node` raises it

kosack, 2021-12-06:
Either should be fine - either auto-remove the slash in TableWriter, or add it in read_table.  Probably the latter is preferred, since it's the user-facing API, whereas TableWriter is quite low-level and users should know what they are doing.

maxnoe, 2022-03-30:
This was fixed in #1817

closed 2022-03-30 by maxnoe