cta-observatory/ctapipe #1828: CameraGeometry position_to_pix_index returns -1 for nan input

moralejo, 2022-01-25:
https://github.com/cta-observatory/ctapipe/blob/39762693534f7dc9b884684c3f82e153bd9da560/ctapipe/instrument/camera/geometry.py#L859

For example, of one uses it on image centroid coordinates to make a "pixelized" 2d histogram, all non-reconstructed images end up in the highest-index pixel (=-1). It would be better to return np.nan.

kosack, 2022-01-27:
I'd call that a bug since the index -1 has a very explicit meaning in python, and does not mean "not found".   However, this function returns an int, not a float, so NaN will not work.  Maybe an exception like `InvalidPosition`,  since in any case the caller needs to handle the failure case?

kosack, 2022-01-27:
Another option would be to return a masked array

kosack, 2022-01-27:
Returning the ids and mask sounds like the best option then, as it forces the user to remember that there are undefined values, and since they probably won't read the documentation! :-)

moralejo, 2022-01-28:
> 
> Could also change the invalid index to something like -interger_max (-9223372036854775808) to be really sure they aren't used as indices...

That sounds like the simplest solution to me, it will force the user to check the index range.

maxnoe, 2022-01-28:
> x = np.array([1.,2.,3.,4.,5.])
> ind = np.ma.masked_array([1,2,2,-1,2], mask=[False, False, False, True, False])
> print(x[ind])
> [2. 3. 3. 5. 3.]

That seems worthy of a report to numpy I think....

moralejo, 2022-01-28:
Seems to be a long-known issue, I wonder whether there is a reason for that
https://stackoverflow.com/questions/3854665/indexing-with-masked-arrays-in-numpy

moralejo, 2022-01-29:
Another issue I just noticed is that position_to_pix_index yields an error in case it gets an empty array. This is a problem e.g. when passing an array with a filter that removes all events. It would be better to return an empty array, right?

PR #2397 (2023-10-25): Use intmin for invalid pixel positions, allow empty arguments
Fixes #1828
closed 2023-10-25 by maxnoe