cta-observatory/ctapipe #1832: CameraDisplay's pixel picker broken

kosack, 2022-01-27:
**Describe the bug**
Clicking on a pixel displays the highlighted pixel in the wrong position for CHEC camera, and clicking pixels stops working if the CameraDisplay is zoomed.

**To Reproduce**
Steps to reproduce the behavior:
```py3
geom = subarray.tel[35].camera.geometry  # a CHEC camera
disp = CameraDisplay(geom)
disp.enable_pixel_picker()
```
then click a pixel

**Expected behavior**
Pixel should be highlighted, and a message "clicked pixel XX" written to the console by default (user overridable to do something useful). 
Additionally, the clicking of pixels should still work when one zooms in on the CameraDisplay, as it used to work.

**Supporting information**
Broken completely for CHEC:
[image]

Highlighting works for LSTCam, but can no longer click after zooming:
[image]

**Additional context**
This feature is used in interactive displays used in test benches

maxnoe, 2022-01-27:
The first issue comes from merging #1819 which switched the patch from a `Rectangle` to a `RegularPolygon` for square pixels, making the subtraction of the pixel_width wrong.

maxnoe, 2022-01-27:
Fix here: https://github.com/cta-observatory/ctapipe/pull/1833

maxnoe, 2022-01-27:
I cannot reproduce the problem with zooming though.

Note: you have to deselect the zoom tool / magnifying glass to be able to click on pixels again.

closed 2022-02-01 by maxnoe