cta-observatory/ctapipe #1876: Reconstructors do not actually use the QualityQuery

maxnoe, 2022-04-12:
**Describe the bug**
Using the `ShowerProcessor`, the event is only checked if it contains at least three (see #1875) telescope events that meet the criteria, however, the `HillasReconstructor` then proceeds to use all telescope events, not only those which pass the quality criteria.

**To Reproduce**

Code in `ShowerProcessor`:
https://github.com/cta-observatory/ctapipe/blob/0b17d4cda17153aec505a119a4aaca1994fdfac4/ctapipe/reco/shower_processor.py#L92-L102

Code in `HillasReconstructor`:
https://github.com/cta-observatory/ctapipe/blob/0b17d4cda17153aec505a119a4aaca1994fdfac4/ctapipe/reco/hillas_reconstructor.py#L191-L195

**Expected behavior**

Only use the telescope events meeting the criteria for stereo reconstruction

kosack, 2022-04-12:
That is a pretty big bug!  I htink this came from refactoring both `HillasReconstrutor` and creating `ShowerProcessor` at the same time with several PRs...   A few possible solutions:

1. write the results of the quality criteria to the event, and then use those to select  in the HillasReconstructor (would also solve #1848, as then the results also get recorded).  But not sure where to put that info (telescope-wise reconstruction)
2. we move this qualityQuery inside HillasReconstructor, which makes some sense, as it is very specific to selecting what HillasReconstructor is able to process.  Other reconstruction methods may have different sets of critieria, e.g. ImPACT can use clipped images, etc.
3. don't call HillasReconstructor.__call__(), and use predict() as before, but that is a bit ugly

maxnoe, 2022-04-12:
Yes, I already started with 2

commit 80b91aa3 (2022-04-12): Fix ShowerProcessor quality checks and make HillasIntersection usable

* Unify API of HillasIntersection and HillasReconstructor
* Enable usage of both in ShowerProcessor
* Give each their own ShowerQualityQuery, so that only passing telescope
  events are used for reconstruction, fixes #1876
commit f03b56e7 (2022-04-14): Fix ShowerProcessor quality checks and make HillasIntersection usable

* Unify API of HillasIntersection and HillasReconstructor
* Enable usage of both in ShowerProcessor
* Give each their own ShowerQualityQuery, so that only passing telescope
  events are used for reconstruction, fixes #1876
closed 2022-04-13 by maxnoe