cta-observatory/ctapipe #1881: Error message for non-existing config-file is suboptimal

maxnoe, 2022-04-19:
**Describe the bug**
Due to the new support for multiple config files, the error message for non-existing files is a bit verbose / confusing:

```
❯  ctapipe-process --config this/does/not/exist
2022-04-19 11:08:18,094 CRITICAL [ctapipe.ctapipe-process] (application.inner): Bad config encountered during initialization: The 'config_files' trait of a ProcessorTool instance contains  which expected <traitlets.traitlets.List object at 0x7f395f9dcd30>, not the str 'Path "/home/maxnoe/this/does/not/exist" does not exist'.
```
**Expected behavior**

Clear error message like "configuration file "<file>" does not exist"

kosack, 2022-04-19:
I guess this is because the `config_files` `Path` trait fails first, as it checks the existence before any config loading happens.  We could make a sub-class called `ConfigPath` perhaps that outputs a more useful error message?

kosack, 2022-04-19:
Ah, but it's also complaining that it is not a list... So it should somehow accept both a list and a single item.  But I thought that should work, as the latest traitlets allows lists to be specified using multiple of the same options?

kosack, 2022-04-19:
I guess it's also the fact that this is a `List(Path)` trait, not just `Path` - not sure why it complains that it is not a List though, since that should not be a problem as it should accept it as a single-item list.

maxnoe, 2023-03-08:
Just ran into this again, really, really misleading and annoying. Need to fix

maxnoe, 2023-03-08:
```
from ctapipe.core import Tool, run_tool

class TestTool(Tool):

    def setup(self):
        pass

    def start(self):
        print(self.config)

if __name__ == '__main__':
    run_tool(TestTool())
```

results in:

```
❯ python test_tool_config.py -c i_do_not_exist.yaml
{}
```

no error, just an empty config.

maxnoe, 2023-03-08:
I have a hard time debugging this, since it never actually seems to enter ctapipe code, the values seem to be lost before in traitlets (e.g. `Path.validate` is never called on the given path)

PR #2591 (2024-09-11): Fix error messages for non-existing path
Fixes #1881 

Now:

```
❯ ctapipe-process --config /does/not/exist
2024-07-15 17:27:04,877 CRITICAL [ctapipe.ctapipe-process] (application.inner): Bad config encountered during initialization: The 'config_files' trait of a ProcessorTool instance contains a Path of a List which expected a pathlib.Path or non-empty str for an existing file, not the PathError '/does/not/exist': does not exist.
```
closed 2024-09-11 by kosack