cta-observatory/ctapipe #1891: Write out `true_image_sum` also when not writing out `true_image`.

maxnoe, 2022-04-22:
**Please describe the use case that requires this feature.**
Currently, the `true_image_sum` is only written when `DataWriter.write_true_images=True`

This is probably the opposite of what we want...

**Describe the solution you'd like**

Write out sum always, full image only when requested.

maxnoe, 2022-04-22:
The question is, where this should be stored.

If we just add it to the `/simulation/event/telescope/images/tel_XXX` tables, we would need do adapt the merge tool a bit to make `skip-images` working (instead of just copying / appending the table nodes, we'd had to modify the tables to exclude the true images  column).

Adding a new table for just this single parameter seems a bit wasteful, but there might be other simulation truth values that are per telescope? Anything that comes to mind that we also want to add?

jsitarek, 2022-04-22:
since this is not a real 'image' parameter, so I would not include the sum there, especially because of the problem that you mention.

one possibility is to put it with the corsika parameters in  SimulatedShowerContainer. Since number of true p.e. is a very basic parameter I think it would also fit there, even if it is obtained only after sim_telarray. 

In MARS we keep also information about how many photons/p.e. survive different stages of the telescope simulation, this is reported in the log of sim_telarray per run but I do not think it is computed (and stored) per event. Those are very technical either way, and I'm not sure if they will be useful (contrary to the total number or p.e.). 

One thing that is very useful is to know how much p.e. come from different types of particles  (e+-/ muons, hadrons)  - this information is available in Corsika, but I do not think it is used in sim_telarray, so again nothing to be stored. 

We store core_x/y position, but not the actual impact w.r.t. the telescope, this would be good to be computed already at the first reading of sim_telarray files. 

I'm not sure if there is some extra information about the trigger available in the sim_telarray outputs - i.e. which pixels gave the trigger, if so this might be useful as well.

maxnoe, 2022-04-22:
> one possibility is to put it with the corsika parameters in SimulatedShowerContainer.

No, that is per array event. We have a parameter here that is per telescope event.

maxnoe, 2022-04-22:
> One thing that is very useful is to know how much p.e. come from different types of particles (e+-/ muons, hadrons) - this information is available in Corsika, but I do not think it is used in sim_telarray, so again nothing to be stored.

This is only available in corsika if used with the `STORE_EMITTER` option, I don't know for sure but I don't think there are aggregated values in simtel array if that option is used and I am not aware any of the standard CTA simulations using this option

maxnoe, 2022-04-22:
> since this is not a real 'image' parameter, so I would not include the sum there, especially because of the problem that you mention.

Yes, agreed, i don't want to include it with the "parameters" which are image parameters calculated on the true image. This is why we could either include it with the images themselves (making the merge logic a bit more complex but I think I figured it out) or add a third table in `simulation/event/telescope`

jsitarek, 2022-04-22:
if it goes as a new table in simulation/event/telescope it can then include two members: true_pe, and true_impact

maxnoe, 2022-04-22:
> true_impact

True impact is not really needed I'd say, since this is in principle there from the shower corex/y and the subarray telescope positions.

Could be convenient though.

jsitarek, 2022-04-22:
sure, it can be computed from those values, but the calculations are not trivial and require also digging out the telescope positions from the array description (and making sure that everything is in the same reference frame) 
since it is really useful, and  it is just one number per telescope I think it is really worth to keep it stored

maxnoe, 2022-04-22:
Implementing this I realized that this information was available in prod3b already (which didn't include the pixel-wise true pe but the total pe).

maxnoe, 2022-04-22:
I implemented the solution that doesn't change the data model for now here #1893, but we should probably change this when also including impact or other parameters we'd want to include to write a third table instead, but this would be a breaking change for the data.

PR #1893 (2022-06-22): Store true_image_sum also when true image is not written
This will also keep the true_image_sum in the merge tool, even when `--skip-images` is given.

Fixes #1891
closed 2022-06-22 by maxnoe