cta-observatory/ctapipe #1892: Store telescope impact distance in simulation container

maxnoe, 2022-04-22:
sure, it can be computed from those values, but the calculations are not trivial and require also digging out the telescope positions from the array description (and making sure that everything is in the same reference frame) 
since it is really useful, and  it is just one number per telescope I think it is really worth to keep it stored

_Originally posted by @jsitarek in https://github.com/cta-observatory/ctapipe/issues/1891#issuecomment-1106439147_

maxnoe, 2022-04-22:
In what frame do we want this information? I guess it's more useful in the tilted frame?

jsitarek, 2022-04-22:
impact is defined as the distance of the telescope from the shower axis (in a sense of a mathematical distance of a point from a line, so measured in the perpendicular plane), so yes, it is easiest to compute it in the tilted frame

maxnoe, 2022-04-22:
> (in a sense of a mathematical distance of a point from a line, so measured in the perpendicular plane)

That is probably the easier way to calculate it, instead of going through the coordinate transformations. Will habe a look.

kosack, 2022-04-25:
We currently compute it in the TiltedFrame aligned with the pointing direction.  I think really it should be in a ShowerFrame (how a person riding on each shower sees the array), but the difference is probably within reconstruction error for sources in the FOV (assuming small angle between the point of origin and the center of the field of view).

kosack, 2022-04-25:
I should mention in protopipe we do compute and store this right now, and it is quite useful to do so to avoid a complicated computation later which is not so easy to add as a simple "feature" in the machine learning steps.  So I would agree to add it to both the SimulationContainer and ideally somewhere in the reconstructed DL2/telescope/geometry data as well.

moralejo, 2022-04-25:
> We currently compute it in the TiltedFrame aligned with the pointing direction. I think really it should be in a ShowerFrame (how a person riding on each shower sees the array), but the difference [...]

Definitely it has to be relative to shower axis, and I think the difference could be pretty large even within FoV, because of distance to shower is way larger than the radius of the light pool.

kosack, 2022-04-25:
Yes, I think you are right that we should do it properly.  This implies we should just make a `Component` to compute it that runs after all Reconstructors.

kosack, 2022-04-25:
something like:
```py3
   impact[tel_id] = np.cross(sky-ground, sky-tel[tel_id])/np.linalg.norm(sky-ground)
```
where sky, ground, and tel are all 3d points

kosack, 2022-04-27:
I made a PR above.  There is still a bug since I did it two ways and they don't agree at certain azimuth angles, so if anyone sees the error, let me know (could also be a bug in the TiltedFrame transforms).

However, assuming it will work correctly, the question now is where to put the pre-computed telescope-wise impact parameters? They could go:

1. in their own table:  */dl2/event/telescope/impact*   and  */simulation/event/telescope/impact *
2. in /*dl2/event/subarray/geometry* as a vector column (not so easy to use it in machine learning then, however)
3. somewhere else?

Any ideas?

maxnoe, 2022-04-27:
Like I said here, at least for the true impact and true image sum I would add a telescope table group at `/simulation/event/telescope/<???>` 

I didn't yet come up with a good name, since parameters already refers to the image parameters as calculated on the true image.

kosack, 2022-04-27:
Agreed that makes the most sense. It's a parameter that requires either a reconstruction or a simulated shower, so we could put it in it's own table like `/{simulation,dl2}/event/telescope/impact/tel_XXX`:

```py3
class ImpactParameterContainer(Container):
    impact_distance =  Field(...)
    impact_distance_uncert = Field(...)
```

PR #1899 (2022-06-26): compute impact distance to shower axis
Creating some functions to correctly compute and store impact distance from the shower axis to the telescopes on the ground.  Fixes #1892 

To do: 
* [x] tests for reading from events
* [x] add to ShowerProcessor (or make a Component)
* [x] figure out where to put this in the output for both reco and simulation 
* [x] update `TableLoader` to read and merge this information
* [x] ~~update `ctapipe-merge` to merge this information~~ → deferred to another PR...

For the output, it could be an array column in the  shower-wise ReconstructedShowerContainer in ` [... cut]
closed 2022-06-26 by maxnoe