cta-observatory/ctapipe #1902: PointingInterpolator

kosack, 2022-04-27:
**Please describe the use case that requires this feature.**

Pointing information is *monitoring data* stored on timescales on the order of 1 Hz.  Therefore it must be interpolated to an event time to get the pointing direction of an event.  This is needed both in the `ctapipe-process` event-loop, as well as useful to have in `TableLoader` to be able to work with interpolated event positions.

There are two types of pointing information:
* the pointing **direction** (azimuth, altitude)
* the pointing **corrections** (dx, dy, camera rotation, maybe skew...) note  above the DL1 level, probably the offsets dx,dy are already applied to the direction, but rotation and skew is still important.

Interpolation in spherical coordinates also has some caveats, as there are poles in the Alt/Az system and these are circular coordinates, so wrapping has to be considered (in case the instrument doesn't return azimuth values > 360 deg for example). 

**Describe the solution you'd like**

a `PointingInterpolator` class that can be used to interpolate a single or multiple event times and returns an event-wise pointing direction.  Something like the following:

```py3
interpolate_pointing = PointingInterpolator(pointing_table, pointing_correction_table)

# given an event table "events"
events['ra_pnt'], events['dec_pnt'], events['rot_pnt'] = interpolate_pointing(events['time'])
```

Inside, would just be a `scipy.interpolate.interp1d` or similar, but taking into account cyclic boundary conditions.

**Additional context**

This could also be automatically constructed in TableLoader, see #1900

maxnoe, 2022-04-27:
See: https://github.com/cta-observatory/ctapipe_io_lst/blob/0b30e9765725a6af33e0da03eef20595de9bcb7d/ctapipe_io_lst/pointing.py#L22

kosack, 2022-04-27:
Looks like the LST version does both bending corrections and pointing interpolation at the same time (which is good).  I don't seem to see it doing any rotation corrections, though - is that not an issue with the LST? (bending that causes the camera to rotate slightly which needs to be applied to the `hillas_phi` (or pixel positions for ImPACT)?

maxnoe, 2022-04-27:
@kosack That's what we have as information now, I don't know what else is planned exactly (I remember talk about star guider pointing corrections, but this not yet used in analysis)

kosack, 2022-04-27:
In any case, having a single class the both interpolates the Pointing Monitoring table and applies the Pointing Correction (if it exists, for now we can use a null operation) would be nice.  Then that allows us to swap in more complex corrections if needed. it's configurable inputs can just be the filename of the H5 file containing the table which could be the same filename as the input, or an external file.

closed 2024-04-09 by maxnoe