cta-observatory/ctapipe #1955: rename ML reconstruction classes

kosack, 2022-06-27:
## Problem
Right now the hierarchy in the `ml`  module is:

```
Reconstructor
     - RegressionReconstructor
         - EnergyRegressor
     - ClassificationReconstructor
         -  ParticleIDClassifier
```

This has two problems:
1. This may cause problems as we also have a `Reconstructor` base class in the `reco` module (and traitlets uses class names to set parameters). 
2. It is better to have the base class in the name of the sub-class, so it's clear to a reader that it is a subclass (when possible)

## Suggested solution:

rename `ml.Reconstructor` → `ml.Predictor`?  (could also see if there is a better word..)

Then you have something like:

```
Predictor
    - RegressionPredictor
        - EnergyRegressionPredictor
    - ClassificationPredictor
        - ParticleClassificationPredictor 
```

maxnoe, 2022-06-27:
Sounds good to me

maxnoe, 2023-10-21:
The class hierarchy was changed before #1767 was merged

closed 2023-10-21 by maxnoe