cta-observatory/ctapipe #2048: Merging of simulated impact distance / merging of telescope DL2

nbiederbeck, 2022-08-03:
Not sure whether this is a Bug or a Feature Request or both

1) We currently don't merge the simulated telescope impact parameters `/simulation/event/telescope/impact`. I think this is a quick fix.

2) We don't merge anything other DL2 than `dl2/event/subarray/geometry`. This includes (not merging) e.g. `HillasReconstructor_tel_distance`, which gets calculated in the most basic DL2 "pipeline" of only Hillas Reco.

kosack, 2022-08-03:
This is probably not a feature but a bug! (or a missing feature rather!)

kosack, 2022-08-03:
Ah, that is also a bug... The Container has a prefix of "impact" so the intention was that the column name is: `impact_distance` not `tel_distance`.   Not sure why that changed?

nbiederbeck, 2022-08-03:
I added a PR that will fix this (in addition to #2049) and created a new issue for the naming bug.

closed 2022-08-25 by kosack