cta-observatory/ctapipe #2057: Processing from DL1 with allowed_tels is messed up

nbiederbeck, 2022-08-16:
*Disclaimer: I will update this description once I have a reproducible example and narrowed down the cause*

**Describe the bug**
We were calculating the angular resolution for some subarrays. 
1) Simtel(allowed_tels) -> DL1 ; DL1(all tels) -> DL2 works (ang-res of 0.5)
2) Simtel(all tels) -> DL1 ; DL1(allowed_tels) -> DL2 fails (ang-res of 5).

My guess is that in the second case the reading is buggy and the true-reconstructed association is wrong.

**To Reproduce**
*Will update*

**Expected behavior**
It should not make a difference where in the pipeline telescopes get selected

nbiederbeck, 2022-08-17:
I created this test that fails with a given file, but with our current test files I cannot reproduce this:
```python

def test_read_from_simtel_and_dl1(prod5_proton_simtel_path, tmp_path):
    """In #2057 reading a subset of allowed tels from
    simtel yields another result as reading from DL1,
    which was produced with all tels.

    This test has three steps:
    1) Create a DL2 file from simtel.
    2) Create a DL1 file from simtel.
    3) Create from that DL1 file another DL2 file.

    Keep in mind that both DL2 allowed_tels need to be the same,
    but different from the allowed_tels in the simtel->DL1 step!
    """
    from numpy.testing import assert_allclose

    from ctapipe.core import run_tool
    from ctapipe.io import TableLoader
    from ctapipe.tools.process import ProcessorTool

    input_path = "/cephfs/projects/CTA/simulations/simtel/Prod5/LaPalma/NSB1x/gamma-diffuse/00xxx/gamma_20deg_0deg_run954___cta-prod5b-lapalma_desert-2158m-LaPalma-dark_cone10.simtel.zst"
    allowed_tels_lapalma_alpha = [
        f"--EventSource.allowed_tels={i}"
        for i in (1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 19, 35)
    ]
    allowed_msts_lapalma_alpha = [
        f"--EventSource.allowed_tels={i}" for i in (5, 6, 7, 8, 9, 10, 11, 19, 35)
    ]

    # 1) Create DL2 from simtel.

    dl2_from_simtel = tmp_path / "from_simtel.dl2.h5"
    argv = [
        f"--input={input_path}",
        f"--output={dl2_from_simtel}",
        "--write-showers",
        "--write-images",
        "--progress",
        "--EventSource.focal_length_choice=EQUIVALENT",
    ] + allowed_msts_lapalma_alpha
    assert run_tool(ProcessorTool(), argv=argv, cwd=tmp_path) == 0

    # 2) Create DL1 from simtel.
    dl1_from_simtel = tmp_path / "from_simtel.dl1.h5"
    argv = [
        f"--input={input_path}",
        f"--output={dl1_from_simtel}",
        "--write-showers",
        "--write-images",
        "--progress",
        "--EventSource.focal_length_choice=EQUIVALENT",
    ] + allowed_tels_lapalma_alpha
    assert run_tool(ProcessorTool(), argv=argv, cwd=tmp_path) == 0

    # 3) Create from that DL1 file another DL2 file.
    dl2_from_dl1 = tmp_path / "from_dl1.dl2.h5"
    argv = [
        f"--input={dl1_from_simtel}",
        f"--output={dl2_from_dl1}",
        "--write-showers",
        "--write-images",
        "--progress",
        "--EventSource.focal_length_choice=EQUIVALENT",
    ] + allowed_msts_lapalma_alpha
    assert run_tool(ProcessorTool(), argv=argv, cwd=tmp_path) == 0

    args = dict(load_dl2=True, load_simulated=True)
    with TableLoader(dl2_from_simtel, **args) as loader:
        events_from_simtel = loader.read_subarray_events()
    with TableLoader(dl2_from_dl1, **args) as loader:
        events_from_dl1 = loader.read_subarray_events()

    # works
    assert_allclose(
        events_from_simtel["HillasReconstructor_alt"],
        events_from_dl1["HillasReconstructor_alt"],
    )

    # regression test, this fails currently
    assert_allclose(
        events_from_simtel["true_alt"],
        events_from_dl1["true_alt"],
    )
```

maxnoe, 2023-01-03:
@nbiederbeck Is this issue still there? Could you pin-point the reason?

maxnoe, 2023-01-03:
I think the problem is that the `HDF5EventSource` ignores events where none of the `allowed_tels` have triggered but the merge tool includes only copies tables as a whole. As a result the file produced with `ctapipe-process` with allowed_tels set will have the "correct"  shower table but the one produced with `ctapipe-merge` will contain additionally the events for which none of the telescopes triggered.

maxnoe, 2023-01-04:
Adding this to the current merge tool will be somewhat complex I think, which leads me to leaning that we maybe should remove `allowed_tels` support from the merge tool.

It is something not really forseen in the current workflow anyway. For processing simulations, we'd likely do:

* process simulations *without* selecting subarray to DL1 images and/or parameters
* merge a number of files so the size makes sense for the DIRAC processing
* run to dl2 for multiple subarrays / settings  (allowed_tels happens here)
* merge to larger training / eval files

So maybe the safest option for now would be to just drop `allowed_tels` from the merge tool.

kosack, 2023-01-04:
> which leads me to leaning that we maybe should remove allowed_tels support from the merge tool.

I think this is the right thing to do: there is no real need to support dropping telescopes during merging.

nbiederbeck, 2023-01-04:
While I agree with your points on the MergeTool, this issue came up without the MergeTool.   Just setting allowed-tels for the ProcessTool at different stages of the pipeline yields this issue.  Some time ago you pointed me to this line https://github.com/cta-observatory/ctapipe/blob/984caa3dc1b42c0d00eecbbd644ad53da907ec7a/ctapipe/io/hdf5eventsource.py#L571, but I could not yet look into this.  I think the MergeTool issue you are talking about is an *additional* problem.

nbiederbeck, 2023-01-04:
See the first comment where I gave an example: this does not work with our test files, but if you provide any simtel-file for `prod5_proton_simtel_path` you should be able to reproduce the issue.

closed 2023-01-13 by maxnoe