cta-observatory/ctapipe #2062: Hillas calculation failed for some telescopes in camera-frame

TjarkMiener, 2022-09-05:
**Describe the bug**
I'm using the latest ctapipe version (v0.16.0) with a subsample of Prod5b. When running the process tool, the Hillas calculation failed (either zeros or nan) for some telescopes in camera-frame.

**To Reproduce**
Steps to reproduce the behavior:
1. `$ ctapipe-process -i /data3/users/nieto/Prod5b_sample/Prod5b_LaPalma_AdvancedBaseline_NSB1x_gamma_North_20deg_DL0_1e6evt/gamma_20deg_0deg_run126___cta-prod5b-lapalma_desert-2158m-LaPalma-dark.simtel.zst --progress --write-images --write-parameters --write-showers --camera-frame`

**Expected behavior**
Hillas calculation should be also working when camera-frame is selected.

**Supporting information**
Screenshot with corrupted tables for tel 3 (LST) and tel 6 (MST with NectarCam):
[image]
Provenance log:
[ctapipe-process.provenance.log](https://github.com/cta-observatory/ctapipe/files/9488873/ctapipe-process.provenance.log)

**Additional context**
It would be also great to store the information about the camera-frame in the meta data and just keep one naming convention for the parameter tables, i.e. removing the prefix "camera_frame_"

maxnoe, 2022-09-08:
I suspect the problem is that for these telescopes the first event is not reconstructible, we then provide the wrong default container and that messes up all following events.

Let me check

maxnoe, 2022-09-08:
Yes, this is the case. We return the global `DEFAULT_IMAGE_PARAMETERS`, which always contains the `telescope_frame` variant. 

I'd solve this by just going ahead and removing the `--camera-frame` option to be honest and not add further complication for something we do not intend to support anymore.

maxnoe, 2022-09-08:
On the other hand, the fix wasn't too hard...

commit 428d1146 (2022-09-08): Merge pull request #2070 from cta-observatory/fix_camera_frame_params

Fix wrong container used for camera frame in ImageProcessor, fixes #2062
commit b377c00a (2022-09-08): Fix wrong container used for camrea frame in ImageProcessor, fixes #2062
closed 2022-09-08 by maxnoe