cta-observatory/ctapipe #2063: Stable optics and camera indices

TjarkMiener, 2022-09-05:
**Please describe the use case that requires this feature.**
Minor stylistic request to improve consistency between runs of one production. 

**Describe the solution you'd like**
Define an order (i.e. linked to tel ids)

**Additional context**
[image]

maxnoe, 2022-09-05:
> ? I thought in python 3.6+ dicts and sets were ordered, but I may be wrong about sets.

Only dicts

kosack, 2022-09-06:
I still don't quite see why files with the exact same subarray in them would end up with these keys in a different order - are we sure the example files at the top of this issue have the same subarray description? If not, then the order would not be expected to be the same. 

As @maxnoe  said, nobody should rely on it being the same either, as it is an internal mapping to build the SubarrayDescription, not something the user should even care about.

maxnoe, 2022-09-06:
@kosack if you want to know the CPython implementation detail that causes this, it is two things:

1. set iteration order happens to be ordered by hash
2. string hashes are randomized between python interpreter sessions:
```
❯ python -c 'print(hash("Hello, World!"))'
7918248467824148683
❯ python -c 'print(hash("Hello, World!"))'
-2028356294309276154
❯ python -c 'print(hash("Hello, World!"))'
2766964094343225856
```

kosack, 2022-09-06:
Then I think it is a good idea to sort first - having the output change slightly between runs with the same input is certainly unexpected.

commit 77afde00 (2022-09-08): Sort unique cameras/types/optics to generate stable indices in hdf5, fixes #2063 (#2066)

* Sort unique cameras/types/optics to generate stable indices in hdf5

* Update docstrings
closed 2022-09-08 by kosack