cta-observatory/ctapipe #2071: Change API of `Reconstructor` and `ShowerProcessor` to allow multiple reconstructions and common predictions

maxnoe, 2022-09-08:
**Please describe the use case that requires this feature.**

Currently, the `ctapipe.reco.Reconstructor` baseclass has the API `__call__(self, event: ArrayEvent) -> ReconstructedGeoemtryContainer`.

This is inflexible, as it makes it impossible to have reconstruction algorithms that directly predict a combination of geometry, energy and particle type, like ImPACT, machine learning, etc.

Also, `ShowerProcessor` only allows a single algorithm.

**Describe the solution you'd like**

Change the API of the general base `Reconstructor` to be `__call__(self, event) -> None`, filling any dl2 container it generates into the event structure.

Change `ShowerReconstructor` to allow a list of `Reconstructor` base classes that are **executed in order**.

I.e. the config should look like this:

```
ProcessorTool:
    ShowerProcessor:
        reconstructors: 
            - HillasReconstructor
            - HillasIntersection
            - EnergyRegressor
            - ImPACT
            - ParticleIDClassifier
        ParticleIDClassifier:
             model_path: ...
        ImPACT:
             template_path:  
             geometry_seed: HillasReconstructor
             energy_seed: RandomForestRegressor 
```

This would still (due to how traitlets works) have the limitation of not being able to e.g. use different configurations of the same reconstructor but would be a big step forward

**Describe alternatives you've considered**

**Additional context**

kosack, 2022-09-12:
The original plan we discussed in the f2f meeting was to separate Reconstructors and AdvancedReconstructors (those that need a previous reconstructor as seed input, like ImPACT and the 3D Model).  With this list method we might not need that, as the order is clear, and it would be up to the user to ensure e.g. that ImPACT comes after HillasReconstructor.    If so, we should ensure there is some automatic check that if an AdvancedReconsructor is in the list, that the required "seed" reconstructors are earlier in the list. 

Or we could stick with that separation to be more robust to accidental misordering.

kosack, 2022-09-12:
> Change the API of the general base Reconstructor to be __call__(self, event) -> None, filling any dl2 container it generates into the event structure.

might be better/cleaner to have reconstructors return the containers only, and let ShowerProcessor fill them into the correct event structure in a common way. This allows them to be used without a full event chain, which makes testing and external use simpler, and it means the logic for filling the event is not spread over each implementation.

commit 4905dc1d (2022-09-23): Merge pull request #2073 from StFroese/shower_processorAPI

Improve ShowerProcessor API, fixes #2071
closed 2022-09-23 by maxnoe