cta-observatory/ctapipe #2078: Editable install broken with newer versions of setuptools

maxnoe, 2022-09-21:
**Describe the bug**
A clear and concise description of what the bug is.

After running `pip install -e .`, I get 

```
$ ctapipe-process --version                                                                                                                   
Traceback (most recent call last):
  File "/scratch/mnoethe/envs/cta-dev/bin/ctapipe-process", line 5, in <module>
    from ctapipe.tools.process import main
  File "/net/nfshome/home/mnoethe/ctapipe/ctapipe/tools/process.py", line 9, in <module>
    from ..calib import CameraCalibrator, GainSelector
ImportError: cannot import name 'CameraCalibrator' from 'ctapipe.calib' (unknown location)
```

when not running in the ctapipe directory. I suspect this is the same issue I faced as with eventio and the solution was to switch to the `src/` layout.

Doing `SETUPTOOLS_ENABLE_FEATURES="legacy-editable" pip install -e ../ctapipe` also fixes the issue  

**To Reproduce**

Setup a fresh development environment.

kosack, 2022-09-21:
Is this a pip issue, or does `python setup.py develop` do the same thing?

maxnoe, 2022-09-21:
There is now a pep defining editable installs, which is now supported by setuptools, and pip calls this new standardized hook.

I didn't test if python setup.py develop also does the new method or is still the legacy method

maxnoe, 2023-09-08:
Ran into this again. THe current fix is to use legacy compat mode:

```
pip install -e . --config-settings editable_mode=compat
```

The only thing that reliably fixed that in other projects was switching to the `src/` based project layout tree.

closed 2024-02-09 by maxnoe