cta-observatory/ctapipe #2080: Document HDF5 format produced by ctapipe in preparation of the DL2 data release

maxnoe, 2022-09-22:
**Please describe the use case that requires this feature.**

For the planned public release of a machine learning enabling DL2 dataset, we should expand the data model (actuall format!) documentation in the docs so that people can work with it.

Current state:
https://cta-observatory.github.io/ctapipe/data_models/dl1.html#ctapipe-data-model

Source files:
https://github.com/cta-observatory/ctapipe/tree/master/docs/data_models

kosack, 2022-09-27:
There is probably a better way to write this documentation that what I did in this outdated version.  In fact we could probably generate it automatically from a file as all the information is in the metadata (file structure, names of containers in each leaf, and even the contents of each container).

kosack, 2022-10-04:
A while back, I created a notebook that would generate a sort of JSON-schema-like object from a ctapipe output file.  Something similar could be used to auto-generate the documentation.  

The notebook  is here in this gist: https://gist.github.com/kosack/6065c38d3de5ae2dbc274d26061ec8d5 

It generates a JSON file with the structure of the file including metadata, tables, and it collapses all tel_* entries into one.

kosack, 2022-10-04:
for documenting the internal ctapipe data model, I also created a notebook that makes a UML diagram from a base container.  However, it seems to no longer work since we changed how container defaults work (default_factory, etc), so I will need to update it.