cta-observatory/ctapipe #2088: Read and write sim_telarray PixelMonitoring information

maxnoe, 2022-09-30:
**Please describe the use case that requires this feature.**

With Prod6, a new object in sim_telarray provides information about the nsb rate, quantum efficiency, high voltage, current and amplification settings of each pixel.

I am not sure where to put them in the data model, since in reality these are monitoring quantities but here these are simulation values randomly derived from the configuration.

I believe these are fixed to once per run per telescope (I asked Konrad to confirm), so I would put these values in `event.simulation.tel[X].service` ? 

    

**Describe the solution you'd like**

- [ ] Add new container
- [ ] Fill in `SimTelEventSource`
- [ ] Write in `DataWriter`
- [ ] Read in `HDF5EventSource`
- [ ] Merge in `MergeTool`

**Describe alternatives you've considered**
Putting this in monitoring. But I think this is not the correct thing to do, as these seem to be fixed, "truthy" values per run per telescope, not deduced values from some measurement.

**Additional context**

@clara-escanuela said she would have a look into implementing it

maxnoe, 2022-09-30:
Yes, for now these are "once per run" so I'd put them into simulation.service.tel

clara-escanuela, 2022-09-30:
Great! I agree. I would include them in simulation

maxnoe, 2022-09-30:
@clara-escanuela just to note, that this then doesn't belong into the `ArrayEvent` but should be exposed as part of the `SimulationConfiguration` of the `EventSource`.

Also note: this information is only there once per run but only after the first event has been read.

kosack, 2022-10-03:
Yes, since these are only once per simulation and have no time-dependence, they can be Service data.  In real data, of course they would be part of the Monitoring info (as a time-series)

maxnoe, 2022-10-03:
I was referring to `eventio.SimTelFile.pixel_monitorings` being filled. Which happens as part of the first event being read.

kosack, 2022-10-03:
> I was referring to eventio.SimTelFile.pixel_monitorings being filled. Which happens as part of the first event being read.

We have to think about where this info goes both in the EventSource and the output data (see some comments in the related PR #2092 ).   

1. Is this info *input* to the simulation?  Then it I think it would be better in the simulation configuration (so `source.simulation_config` and in the output file `/configuration/simulation/tel/pixel_settings` 
2. is this info *output* from the simulation? (i.e. is it simulated monitoring information that you woudl get from a real telescope?)  In that case it would go in the output in `/simulation/service/pixel_settings`, but not sure where in the EventSource internal representation - so far we have no monitoring or service there, only those that are interpolated to individual events.  I think this info should never change event-to-event, so it probably doesn't go in `event`, but rather in `source.something`

If option 1 is correct, then it would be very user-unfriendly to rely on having an event loaded first, so we should do something to trick the source into reading it (perhaps read an event and rewind on init?)

maxnoe, 2022-10-04:
It's kind of in between 1 and 2: its is true expected values which are randomly generated from the input configuration.

It for sure should not go into `event`, it's a per run property.

Here is part the sim_tel structure:

```
CameraSettings[2002](telescope_id=87)
 CameraOrganization[2003](telescope_id=87)
 PixelSettings[2004](telescope_id=87)
 DisabledPixels[2005](telescope_id=87)
 CameraSoftwareSettings[2006](telescope_id=87)
 DriveSettings[2008](telescope_id=87)
 PointingCorrection[2007](telescope_id=87)
 MCShower[2020](shower_id=1)
 MCEvent[2021](event_id=100)
 TelescopeData[1204](event_id=100)
     PhotoElectrons[1208](array_id=0, telescope_id=0)
     PhotoElectrons[1208](array_id=0, telescope_id=1)
     PhotoElectrons[1208](array_id=0, telescope_id=2)
     PhotoElectrons[1208](array_id=0, telescope_id=3)
     PhotoElectrons[1208](array_id=0, telescope_id=4)
     And 28 objects more of the same type
 PixelMonitoring[2033](telescope_id=1)
 PixelMonitoring[2033](telescope_id=2)
 PixelMonitoring[2033](telescope_id=3)
 PixelMonitoring[2033](telescope_id=4)
 PixelMonitoring[2033](telescope_id=5)
 And 81 objects more of the same type
 CameraMonitoring[2022](telescope_id=1)
 LaserCalibration[2023](telescope_id=1)
 CameraMonitoring[2022](telescope_id=2)
```

So you can see that this info is not part of the configuration block (CameraSettings etc.) *before* the first event but comes *after* the CORSIKA related blocks of the first event and *before* the simtel related data of the first event