cta-observatory/ctapipe #2090: Split-up ctapipe into multiple smaller packages

maxnoe, 2022-09-30:
It now came up in multiple settings that other CTA related projects want to use certain ctapipe functionality but ctapipe itself is too large of a dependency or would in fact introduce a circular dependency (e.g. pyirf needing ctapipe.coordinates, ctapipe needing pyirf for IRFs).

So we might think about moving certain, stable core functionality into their own packages so ctapipe and other packages can easier depend on them.

These come to mind:

* Configuration / Component / Tool (`cta-configsystem` ?)
* Containers / low-level IO for containers `cta-containers` ?)
*  coordinate frame definitions (`cta-coordinates`?)

Note that for `cta-configsystem`, at least longterm we also want to have a system not based on `traitlets` due to the different shortcomings we now know...

HealthyPear, 2022-09-30:
I agree with the idea!
For some modules, the ability to extract them as new packages might depend on their expected scope.

If in this issue we care only about IACTs everything becomes more conservative, my following example might be an extreme case. I am working on a ctapipe "spinoff" for instruments like HAWC and I think it's a nice recent application for this plan.

Taking into account that I use the least amount of code to develop small ingredients (like an EventSource):

- I could recycle the whole `core` module but I had to change things here an there e.g. `tel_id` to `tank_id`, so one might think to call it `unit_id` and do the same for the rest
- `coordinates` was easy, but of course, I ditched everything that is not related to the ground frame
- the explicit `containers` is again something I had to modify but some of them (like EventIndexContainer) are used without modifications (but the container class is in `core`) - this also depends on the data models so not easy to export outside of IACTs - maybe we should export only some of the more basic ones?
- the configuration system might be the only one really easy to recycle

HealthyPear, 2022-09-30:
Also, `ctapipe.vizualization` could go into `ctaplot` (if that is still a thing)

HealthyPear, 2022-09-30:
Classes like `ctapipe.reco.ShowerProcessor` and `ctapipe.reco.Reconstructor` are also instrument-blind until output enters the game

HealthyPear, 2022-09-30:
and also the whole `ctapipe.utils`

I tested it and the only things to change are stuff like default URLs

kosack, 2022-10-03:
This was always a future plan, but I would not like to make the maintenance so much harder just because people complain about installing a 13k line package (which is not very large). Most of the dependency issues are not due to ctapipe itself, but rather because we include things like matplotlib and astropy that pull a lot of deps in their default conda package.  

Multiple packages do solve some problems, but make it also far more complicated to use and maintain ctapipe . As we are already a super small team, I would avoid it for now. Look what happened to e.g. ctaplot, which basically became abandoned. 

This can be on our roadmap for the future, but I think now is premature and will only hamper development.  Most of the package that want to use ctapipe (like the simulation software and calibration pipeline in CTAO) in fact need much of the functionality anyhow

kosack, 2022-10-03:
I think for very large changes to the code, particularly after 1.0, we should institute something like the PEP/APE/PIG system used by python, astropy, gammapy respectively, where we can have formal requests for restructuring with a well-defined review and acceptance process