cta-observatory/ctapipe #2099: Use jupytext for notebooks in the documentation?

kosack, 2022-10-19:
**Please describe the use case that requires this feature.**

Looking at diffs and quickly updating notebooks in the docs currently requires external tools like *nbdime* or *Review Notebook App* due to the fact that the `*.ipynb`  format is JSON-based.  JSON is also not very GIT-friendly.

**Describe the solution you'd like**

jupytext (https://jupytext.readthedocs.io) solves this by using a line-based format for notebooks (basically a python script with special comments to store the metadata). 

It would be nice to use this and migrate the current docs to this format.   

Disadvantage: It would add one more dependency on the jupytext package for documentation building, and would also require anyone using the notebooks outside of the docs to install this plugin.   Also, not sure that GitHub auto-renders jupytext docs, probably not (but that may not be a big problem as the docs are rendered to readthedocs)

vuillaut, 2022-10-21:
I was going to write that an additional potential drawback is the inability to run on mybinder, but it's actually possible:
https://jupytext.readthedocs.io/en/latest/faq.html?highlight=binder#can-i-use-jupytext-with-jupyterhub-binder-nteract-colab-saturn-or-azure
👍

maxnoe, 2023-10-21:
Done in #2377

closed 2023-10-21 by maxnoe