cta-observatory/ctapipe #2100: Move validation of class enum trait to run time instead of import time

maxnoe, 2022-10-20:
**Please describe the use case that requires this feature.**

Currently, we define class traitlets like this:

```
class Foo(Component):
    bar_type = create_class_enum_trait(Bar...).tag(config=True)
```

this runs at import time of Foo. Which makes it impossible to use subclasses of `Bar` that are only imported at a later stage.

This is problematic for subclasses defined in other submodules than the base class and the containing component.

THis was triggered by the machine learning module, where `ShowerProcessor` has a class enum trait of `Reconstructor` but because `ctapipe.ml.sklearn` cannot import `Reconstructor` without also importing `ShowerProcessor` (as `ctapipe.reco.__init__` imports it), the ml reconstructors are not yet defined when the class enum trait is created.

This is even more problematic for the case of plugins. 

**Describe the solution you'd like**

Move the check if a provided string is an available class to the runtime (i.e. the validate method).

StFroese, 2022-10-21:
So this is the reason for this:

https://github.com/cta-observatory/ctapipe/blob/05beb25fb31f27f243f25b4542b1dcea6201300e/ctapipe/reco/__init__.py#L3-L4

maxnoe, 2022-10-21:
Exactly

commit ca7e9dbb (2022-10-28): Add and use ComponentName traitlet instead of create_class_enum_trait, fixes #2100
commit f61cc7aa (2022-10-28): Merge pull request #2102 from cta-observatory/classname_trait

Add and use ComponentName traitlet instead of create_class_enum_trait, fixes #2100, #1874
closed 2022-10-28 by kosack