cta-observatory/ctapipe #2103: Python 3.11 support

LukasNickel, 2022-10-25:
Python 3.11 is out now, we should add it to the CI
(https://www.python.org/downloads/release/python-3110/)

Not a whole lot changed, should be smooth sailing.
Do we remove 3.8 now?

maxnoe, 2022-10-25:
We decided to follow the  nep29 schedule, which means dropping 3.8 in April 2023

kosack, 2022-10-25:
It also usually takes a little while for there to be conda packages built for a new python release, but we could open a PR now and use that to monitor when the tests pass.

maxnoe, 2022-10-25:
conda-forge packages for 3.11 are already available

maxnoe, 2022-10-25:
But I guess it will take a while for numpy and co

maxnoe, 2022-11-26:
The main blocker is numba, tracking issue here: https://github.com/numba/numba/issues/8304

maxnoe, 2023-06-05:
This was done in #2107

closed 2023-06-05 by maxnoe