cta-observatory/ctapipe #2104: Using CameraReadout.from_name crashes with a version conflict error

evertrol, 2022-10-26:
If I use

```
from ctapipe.instrument.camera.readout import CameraReadout

CameraReadout.from_name("ASTRICam")
```

I'll get an error:
```
Traceback (most recent call last):
  File "/some/home/dir/Downloads/testcta.py", line 22, in <module>
    readout = CameraReadout.from_name(camera)
  File "/some/home/dir/.pyenv/versions/3.10.6/lib/python3.10/site-packages/ctapipe/instrument/camera/readout.py", line 157, in from_name
    return CameraReadout.from_table(table)
  File "/some/home/dir/.pyenv/versions/3.10.6/lib/python3.10/site-packages/ctapipe/instrument/camera/readout.py", line 212, in from_table
    raise IOError(
OSError: CameraReadout table has unsupported version: 1.0, supported are: {'3.0'}.
```

Something similar happens with
```
TelescopeDescription.from_name(optics, camera)
```
with appropriate `optics` and `camera` values. It's effectively the same error, because `TelescopeDescription.from_name` calls `CameraReadout.from_name`.

But
```
from ctapipe.instrument.camera.geometry import CameraGeometry

CameraGeometry.from_name("ASTRICam")
```
works fine (as also per the tutorials).

Interestingly, the test `ctapipe.instruments.tests:test_telescope_from_name` uses the `TelescopeDescription.from_name ` as well, and that runs fine. This seems to be, because `test_telescope_from_name` uses the `svc_path` fixture, which it turn calls the `instrument_dir` fixture (both in `ctapipe/instrument/conftest.py`), which through a few more indirections, ends up calling `ctapipe.instrument.camera.readout:to_table(self)`. That last method fixes the metadata for the camera-readout table: `TAB_VER=self.CURRENT_TAB_VERSION`, with the latter hardcoded as "3.0" at the top of the class definition.

This step (calling `to_table(self)` and fixing the version number) doesn't happen when I simply use `CameraReadout.from_name` directly, as evidenced by the error.

I did come across issue #1978, so I'm not sure what the status and plan here is. `.from_name` usage is supported (apparently) for the geometry, but not for the readout? The related PR #1985 focuses on the tests, but the test I have compared with (noted above) still uses `TelescopeDescription.from_name`.

I've filed this as a bug, but I'm not sure what the preferred resolution is, given the linked issue. If it is to be abandoned, perhaps a DeprecationWarning is useful (though that's already too late, since it crashes, instead of warning me). If it is to be removed, how does one obtain the relevant information? And should the relevant test (and perhaps other tests) succeed, with the fixtures hardcoding the version, or actually fail (similar to my direct usage)?

maxnoe, 2022-10-26:
@evertrol I cannot reproduce your error. We did update the files on the data server to have version 3.0.

Might I suggest you try to clear the ctapipe download cache? E.g.: `rm -r ~/.cache/ctapipe`?

evertrol, 2022-10-26:
Hm, interesting.

Had cleaned the cache before, just tried again, and I can see the file being downloaded:
```
Downloading ASTRICam.camreadout.fits.gz: 100%|██████| 2.99k/2.99k
```
Files get downloaded to the subdir `cccta-dataserver.in2p3.fr/data/ctapipe-extra/v0.3.4`, which  matches the URL it is checking for the variations (fits, ecsv etc) of the table.

But I still get the same error with those two lines of code at the top.

I'm using the master branch, 05beb25f, which is the most recent commit afaict.

I'll check on another machine, in a v-env, a bit later.

Perhaps in the meantime, anyone has some quick and easy pointers for me to debug this further?

maxnoe, 2022-10-26:
@evertrol I am sorry, I tried only with `CameraGeometry`, not `CameraReadout`, with that I get the same error!

We will update the files...

But please note that the files on the dataserver are very old. The ASTRICam is the one from prod3 or even from prod2, not sure even!

It is much better to just read the subarray from a simulation file you know how it was produced.

Via `EventSource`:

```
with EventSource("...path/to/file") as source:
    readout1 = source.subarray.tel[1].camera.readout
```

Or if you only want the subarray, you can now use:
```
subarray = SubarrayDescription.read("path/to/file")
readout = subarray.tel[1].camera.readout
```

which will work with all input files supported by ctapipe and plugins

evertrol, 2022-10-26:
Ok, thanks!

I do need (for now) only the geometry files for some cameras, but not from simulations (I don't even have any of those). This seems a good and easy way to obtain them (though I do hope then that the geometry hasn't changed too much compared to e.g. the readout info). The readout and telescope description were just my attempts to see what other information I could obtain. I guess I'll ask around for those, but it does seem nice to have them all in one place.

maxnoe, 2022-10-26:
@evertrol You can use a more recent simulation file from the dateserver like this:

```
subarray = SubarrayDescription.read("dataset://gamma_prod5.simtel.zst")
```

kosack, 2022-10-27:
I uploaded new versions of the tables to the dataserver, which should include the ASTRICam geometry and readout files.  Once #2108 is merged, it should work for you.  If you don't want to update ctapipe, however, you can set the download path to the new one with the updated files at the top of your script:
```python
from ctapipe.utils import datasets
datasets.DEFAULT_URL = "http://cccta-dataserver.in2p3.fr/data/ctapipe-test-data/v1.1.0/"
```

Alternately, you can download the ASTRICam files first into a local directory and set CTAPIPE_SVC_PATH to include that directory:
```sh
mkdir ctapipe_stuff
cd ctapipe_stuff
curl -O http://cccta-dataserver.in2p3.fr/data/ctapipe-test-data/v1.1.0/ASTRICam.camgeom.fits.gz
curl -O http://cccta-dataserver.in2p3.fr/data/ctapipe-test-data/v1.1.0/ASTRICam.camreadout.fits.gz
export CTAPIPE_SVC_PATH=$PWD:$CTAPIPE_SVC_PATH   
```
Then it will work with no change to your scripts, as datasets get looked up first in this directory, and only if not found they are downloaded

maxnoe, 2022-10-27:
@kosack that probably means we should also make the url configurable via env variable...

closed 2022-10-27 by maxnoe