cta-observatory/ctapipe #2111: Make use of --overwrite / -f in tools consistent

nbiederbeck, 2022-11-02:
we have `-f` as "force" in the other tools (including ml tools). But I think we can adress the `--overwrite/-f` in another PR to make everything consistent (and maybe inside Tool?)

_Originally posted by @nbiederbeck in https://github.com/cta-observatory/ctapipe/pull/1767#discussion_r1011531771_

nbiederbeck, 2022-11-03:
Additionally, if we could check the `output_path.exists() and not self.overwrite` for each output file *during initialization*, that would speed things up. Currently, that check is done during `Tool.finish()` when calling `Component.write()`.

kosack, 2022-11-03:
Sounds good. Should definitely change everything to `--overwrite`.  I dislike `--force`, since it is not clear what is being forced.  Astropy and gammapy also use `--overwrite` for everything.  I don't think there has to be a short name for this, and would even discourage adding one, so no `-f` or `-o`, etc.

nbiederbeck, 2022-11-03:
What's your opinion on multiple output files?

I assume one `--overwrite` is enough to overwrite *all* output files.

kosack, 2022-11-03:
Yes, I think so, otherwise it gets quite confusing to have separate options for each.

closed 2023-01-24 by maxnoe