cta-observatory/ctapipe #2114: Joining tables in TableLoader is a performance bottleneck

maxnoe, 2022-11-08:
Applying machine learning models to large files with many telescope events spends most of the time in joining tables in `TableLoader`. 

Connected astropy issue: https://github.com/astropy/astropy/issues/13798

kosack, 2022-11-08:
this [StackExchange post](https://stackoverflow.com/questions/69666319/numpy-argsort-slow-performance) notes using numba parallel version of argsort to gain speed, but since the call to np.argsort is deep in astropy.table, I guess we can't do that.

maxnoe, 2022-11-18:
One possible speedup would be to use join only when joining telescope to subarray tables. At the moment, we use `join` also to combine tables that should have the same structure. 

At the moment, the way datawriter works, these are guaranteed to be identical in practice, which is maybe something we should just enforce, at it is always much more officient to just add columns assuming the index is identical (checking that is also actually less expensive than performing any merge) and make that an invariant of the data model / format.

So all subarray tables should have the same index (at least for triggered events, the list of simulated showers might be an exception as soon as `eventio` is able to provide that), as well as all telescope tables *for the same telescope*

maxnoe, 2022-11-25:
Fixed mostly in #2126 and #2133

closed 2022-11-25 by maxnoe