cta-observatory/ctapipe #2116: General high level API for `Reconstructor`

maxnoe, 2022-11-08:
**Please describe the use case that requires this feature.**

In the current master, the only required API of `Reconstructor` is `__call__(self, ArrayEvent)`. The `SKLearnReconstructors` adds to this the `predict_table` method, which performs monoscopic prediction given a key, which at this time must be the `TelescopeDescription`.

It would be nice to have general interface for `Reconstructor`, so that the `ctapipe-apply-models` and `ctapipe-process` tools can apply all valid subclasses of `Reconstructor` by just being given the config and / or a path to read a pickled reconstructor from.

The problem here is that the `predict_table(key, telescope_table)` is specific to reconstructors predicting on one telescope table at a time, i.e. this is not feasible for e.g. the `HillasReconstructor` or machine learning algorithms directly predicting stereo results.

It also does not work with reconstructors able to predict multiple properties (i.e. ImPACT) at the same time.

**Describe the solution you'd like**

Define the interface of `Reconstructor` so that use cases are covered and `ctapipe-process` can work event-wise with all reconstructors and `ctapipe-apply` can work in bulk with all reconstructors.

One possibility would be to have more methods, that can be implemented by reconstructors.

WIP:

```
class Reconstructor:
    def __call__(self, event: ArrayEvent):
        """As before: predict and fill prediction into the given ArrayEvent"""

    def predict_table(self, table) -> tuple[dict[str, Table], dict[str, dict[int, Table]]]:
        """
        Predict on a table as returned by TableLoader.read_telescope_events()

        Returns
        -------
        array_predictions :
            dict mapping properties to tables of stereo predictions, e.g.:
            ``{"energy": Table(), "geometry": Table()}``
        telescope_predictions :
            dict mapping properties to dicts mapping tel_ids to tables of mono predictions, e.g.:
            ``{"energy": {1: Table(), 2: Table(), }}```
        """
```

The problem here is that this would not work for reconstructors needing images, as a single table cannot hold the differently shaped images in the same table.

**Describe alternatives you've considered**

Not really any.

kosack, 2022-11-08:
We might not really need to have reconstructors like ImPACT be "apply-able", as those can just run in ctapipe-process. It's really only ML-based prediction reconstructions that  need to do a test/train cycle where ctapipe-apply is very useful. Even i a final analysis, the predict can be done in the ctapipe event loop however, once models exist. 

Also, do we expect that the monoscopic ML reconstructor (disp method) will be able to be applied after the fact? It would need to have the pointing info interpolated to each event, but maybe that is ok.  Needs some thought if that is necessary.

maxnoe, 2022-12-16:
Thinking about this again, the case of monoscopic table-wise predictions is pretty easy to define a common high level API.

```
class Property(str, Enum):
    """Enum for the particle properties a reconstructor can predict"""
    ENERGY = "energy"
    ParticleType = "classification"
    DIRECTION = "direction"
    DISP = "disp"

class Reconstructor:
    #: this reconstructor predicts energy and direction
    properties = (Property.ENERGY, Property.DIRECTION)

    def __call__(self, event: ArrayEvent):
        """As before: predict and fill prediction into the given ArrayEvent"""

    def predict_telescope_table(self, key, table) -> dict[Property, Table]:
        """
        Predict on a table as returned by TableLoader.read_telescope_events()

        Parameters
        -----------------
        key: int or TelescopeDescription or str
            Key for chosing the model, a tel_id (int) or TelescopeDescription instance or telescope description str

        Returns
        -------
        telescope_predictions :
            dict mapping properties to dicts mapping tel_ids to tables of mono predictions, e.g.:
            ``{PROPERTY.ENERGY:  Table()}```
        """
```