cta-observatory/ctapipe #2118: ReconstructedGeometryContainer units do not match for HillasReconstructor

StFroese, 2022-11-16:
**Describe the bug**
The ReconstructedGeometryContainer does return angles in rad instead of deg when using HillasReconstructor.

**To Reproduce**
```python
In [2]: from ctapipe.io import EventSource
   ...: from ctapipe.reco import HillasReconstructor
   ...: 
   ...: 
   ...: source = EventSource("/Users/stefan/Downloads/gamma_20deg_180deg_run3___cta-prod6-paranal-2147m-Paranal-dark.dl2.h5", max_events=1)
   ...: events = list(source)
   ...: 
   ...: reco = HillasReconstructor(source.subarray)
   ...: 
   ...: for event in events:
   ...:     reco(event)
   ...: 

In [3]: event.dl2.stereo.geometry['HillasReconstructor'].alt
Out[3]: <Latitude 1.21863615 rad>
```

**Expected behavior**
Should return angle in deg.

```
ctapipe.containers.ReconstructedGeometryContainer:
                           alt: reconstructed altitude with default nan deg
                                [deg]
```

**Solution**
Not sure but I guess the Container just has to be validated.

StFroese, 2022-11-16:
Ok I just saw this comment in the Container validation code: `This is not intended to be run every time a Container is filled, since it is slow, only for testing a first event.`

So in this case maybe just convert the units upfront for HillasReco

maxnoe, 2022-11-16:
The point of having unit information attached is that the units don't really matter as long as they are of the correct dimension.

So yes, it would be nicer to be consistent, but it's not a bug per-se and `Container.validate` checks `self.unit.is_equivalent(value.unit)`, not `self.unit == value.unit`:

```
In [1]: from ctapipe.containers import ReconstructedGeometryContainer

In [2]: import astropy.units as u

In [3]: ReconstructedGeometryContainer(alt=20 * u.deg).validate()

In [4]: ReconstructedGeometryContainer(alt=0.5 * u.rad).validate()

In [5]: ReconstructedGeometryContainer(alt=0.5 * u.m).validate()
...

UnitConversionError: 'm' (length) and 'deg' (angle) are not convertible
```

maxnoe, 2022-11-16:
And in the output files, it is converted to the `Field.unit`.

StFroese, 2022-11-16:
Ah ok I didn't check that

closed 2022-11-16 by StFroese