cta-observatory/ctapipe #2122: Incomplete contributors list

vuillaut, 2022-11-16:
**Describe the bug**

It came to my attention that some code contributors do not appear in the list of contributors provided by GitHub.
Essentially, the output of  
```
git shortlog -sne
```
differs from https://github.com/cta-observatory/ctapipe/graphs/contributors

This is apparently due to differences in emails that people use in their GitHub account and the ones used to commit code (thanks @maxnoe for the explanation).
The issue is that we currently rely on GitHub to provide the list of contributors to Zenodo when doing a release.

Moreover, we might also want to give credit to other contributors that did not commit code but still have made an important contribution.

**To Reproduce**
Steps to reproduce the behavior:
1. `git shortlog -sne`
2. Compare to https://github.com/cta-observatory/ctapipe/graphs/contributors
3. or to https://zenodo.org/record/7118605

**Expected behavior**

All contributors are credited.

**Additional context**

In the context of FAIRness, a complete metadata file should be included in the repository.
We have the same issue in basically all repository.

maxnoe, 2022-11-16:
This problem would be lessened, if GitHub would actually use `.mailmap`, but it doesn't. So yes, we should add a `.zenodo.json` for the zenodo deployment.

@vuillaut you know better what the best way for that is: directly writing zenodo.json? Or using something else and then automatically creating it

maxnoe, 2022-11-16:
For the past zenodo entries, @kosack can also edit the author list, if we wish to make that effort.

vuillaut, 2022-11-16:
@maxnoe 
Ultimately, yes we should write a `.zenodo.json` I think. I am considering different solutions here... 

You are right, there is some kind of hidden magic here, but I don't think it's based on mailmap, I see no reference to mailmap in the zenodo backend code... 🧐

maxnoe, 2023-10-21:
We should add updating `AUTHORS.md` to the release procude documented in the developer guide

kosack, 2024-08-01:
Reopening since this got auto-closed, but there is still the issue of previous releases.  Should we edit one or more of the previous releases to include the correct authors, or just wait for the next release?

kosack, 2024-08-01:
Also, perhaps we can do a CI check like using the git shortlog output and seeing if any entries are not in AUTHORS to avoid forgetting to add new contributors?

PR #2590 (2024-08-01): Update AUTHORS, add to release procedure
Fixes #2122
closed 2025-08-21 by maxnoe