cta-observatory/ctapipe #2127: Add multiplicity explicitly to output containers

maxnoe, 2022-11-20:
**Please describe the use case that requires this feature.**

We put `telescopes` into the stereo output containers, so the information is there in principal, but I think it would be very convenient to also simply store the number of telescopes.

StFroese, 2022-11-22:
So basically this?
```python
AStereoOutputContainer(
...,
telescopes=telescopes,
multiplicity=len(telescopes),
)
```

maxnoe, 2022-11-22:
Yes, or `np.count_nonzero(telescopes)` if telescopes is already the mask, not the list of telescope ids.

StFroese, 2022-11-22:
Ok, great. Since there will be no tel_id 0, this will catch both cases

maxnoe, 2023-09-19:
In another discussion, we decided against this and instead implemented the `multiplicity` function on the `SubarrayDescription` to be used in e.g. quality queries

closed 2023-09-19 by maxnoe