cta-observatory/ctapipe #2135: Machine Learning Tutorial

nbiederbeck, 2022-11-25:
Since we now have the machine learning tools, we need to show users how to work with them.

They should be (at least) [documented with the command line tools](https://cta-observatory.github.io/ctapipe/tools/index.html), but I would much more prefer a [tutorial in the documentation](https://cta-observatory.github.io/ctapipe/tutorials/index.html).

I open this as a new issue, instead of appending at #1755, because this should focus only on documentation and tutorials not new features.

PS: we might also overhaul the whole tutorial section, there may be some outdated stuff and I really like how [Gammapy handles their tutorials](https://docs.gammapy.org/1.0/tutorials/index.html)

kosack, 2022-11-25:
Yes new tutorials have long been on the todo list.  I think we can also separate them into two categories: high-level tutorials (like using the machine learning tools) and API-level tutorials similar to what is there already.  For the high level ones, the primary one would be to show a complete step-by-step analysis, at least up to generating DL2 event lists with geometry, energy, and gammaness.

kosack, 2022-11-25:
This also might be a nice time to look into switching the tutorial notebooks to use Jupytext #2099

StFroese, 2022-11-25:
I think there should also be a data flow diagram showing which tools are applied at which step and which files are or can be generated along the way

closed 2025-03-31 by maxnoe