cta-observatory/ctapipe #2142: Train ML models with different hyper parameters and features

StFroese, 2022-11-28:
**Please describe the use case that requires this feature.**
Train and apply multiple regressors and/or classifiers for different hyper parameters and features.

**Describe the solution you'd like**
Currently the class name of the model is used as a prefix for the keys in the table.
Adding a traitlet for the prefix for these models would allow the user to store the output of multiple ml classes in the table without overwriting the output of other models.

maxnoe, 2022-11-28:
Unfortunately, the config system does not allow having the same class with different configs, so it is at the moment impossible to configure e.g. two `EnergyRegressor` instances in the `reconstructors` list with different settings.

This will thus have to wait until we change the config system. Until then, multiple outputfiles have to be used I think.

maxnoe, 2022-11-28:
#1406

kosack, 2022-11-29:
The config system is something I think will be overhauled in ctapipe-2.x as it is going to be a large change, for 1.x what we have now is fine and we shouldn't delay because of it (we can just run multiple trainings, for example)