cta-observatory/ctapipe #2143: Update Docs and Tutorials

StFroese, 2022-11-30:
Issue for collecting required updates/reworks of the documentation and tutorials. Just add stuff if you think something needs an update.
Initially I will just add every chapter of the docs.

# Updates
Please check a box if you think this needs an update and also leave a comment on why you think this is required.

<details>
<summary>Getting Started For Developers</summary>

- [ ] Get the ctapipe software
- [ ] Developing a new feature or code change
- [ ] More Development help

</details>

<details>
<summary>Development Guidelines</summary>

- [ ] Style Guide
- [ ] Code Guidelines
- [ ] Support Libraries
- [ ] Making and Accepting Pull Requests
- [ ] Maintainer Info
- [ ] For ROOT users

</details>
<details>
<summary>Tutorials</summary>

- [ ] Getting Started with ctapipe
- [ ] Analyzing Events Using ctapipe
- [ ] Coordinates usage in ctapipe
- [ ] Exploring Raw Data
- [ ] Explore Calibrated Data
- [ ] Make a theta-square plot

 
</details>
<details>
<summary>Examples</summary>

- [ ] Basic Image Cleaning and Dilation
- [ ] Use N-dimensional Histogram functionality and Interpolation
- [ ] Convert camera images to pixels on a square grid
- [ ] Working with Instrumental Descriptions
- [ ] Example of a Camera Display
- [ ] Using Container classes
- [ ] Creating command-line Tools
- [ ] Using the ctapipe Provenance service
- [ ] Writing Containers to a tabular format

 
</details>
<details>
<summary>Command-Line Tools</summary>

- [ ] Data Processing Tools
- [ ] Other Tools
 
</details>

<details>
<summary>Frequently Asked Questions</summary>

- [ ] Technical Issues
 
</details>

<details>
<summary>Data Models</summary>

- [ ] CTAPipe Data Model
 
</details>

# New
This lists all new stuff that has so be added to the docs. Link the PRs or issues.

- [ ] Update tutorials that require CameraFrame for image processing #2141 
- [ ] #2135 

# Additional questions
Some questions that came while writing this

- [ ] What about the Change Log in the docs, do we need this?
- [ ] References should be updated if your code comes from a paper (e.g #1996 should do this)
- [ ] Do we need Tutorials and Examples? Seems kind of redundant to split them

kosack, 2022-11-30:
- The main need is to add some new tutorials about running the command-line tools to do a standard analysis.  Nearly all the others need some small updates, but they are mostly ok.  I can help to clean those up. 

- The ChangeLog hasn't been consistently updated - we should switch to using towncrier as mentioned in #1995 

- For the tutorials vs Examples question: the idea were tutorials were higher-level and relate to analysis, while examples were just quick demos of various API features. But if we add a command-line analysis tutorial (or more than one), we  should split the Tutorials section into : _API Tutorials_  and _Command-line Tutorials_,  so the hierarchy is:
  - Command-line and standard analyses (where most users should start)
  - API tutorials (for advanced users)
  - API examples (for developers)

kosack, 2022-11-30:
[Gammapy's documentation](https://docs.gammapy.org/1.0/) is also a good example of how this all could be re-structured to be simpler.

StFroese, 2022-11-30:
Also as mentioned in #2135, we should switch to jupytext as suggested by @kosack

StFroese, 2022-11-30:
`Get the ctapipe software` should be updated with #2087

StFroese, 2022-12-01:
#2144

maxnoe, 2025-08-25:
This is basically a large, not achievable issue. Closing. 

We have multiple smaller issues that can be addressed.