cta-observatory/ctapipe #2165: Move `TableLoader` options from being traitlets to the `read_...` methods

maxnoe, 2022-12-13:
**Please describe the use case that requires this feature.**
A clear and concise description of why this feature is needed. [...]

There is no reason why these should be fixed at the class level, e.g. this should be possible:

```
with TableLoader(path) as loader:
    events = loader.read_telescope_events(images=False, parameters=True, true_shower=True)
    images = loader.read_telescope_events_by_type(images=True, parameters=False, true_shower=False)
```

nbiederbeck, 2022-12-19:
Related: #2012

ccossou, 2023-10-25:
I'm also working on this one, Jonas, I'm interested to work with you, esssentially because I don't necessarily understand the question. 

The example show parameters for images, parameters and true shower, but what about dl1 and dl2? 

The currents traits we have in TableLoader are:
```
load_dl1_images
load_dl1_parameters
load_dl1_muons
load_dl2
load_simulated
load_true_images
load_true_parameters
load_instrument
load_observation_info
```

ccossou, 2023-11-06:
The [branch](https://github.com/cta-observatory/ctapipe/tree/move_table_loader_traits) is working now and pass all updated unit test. Maybe a pull request should be open. What do you think?

One last item I recall that is not yet implemented, is adding unit test to ensure deactivating an option has an effect. @Hckjs since this is mainly your work, I let you decide.

Hckjs, 2023-11-08:
First of all thanks for the help @ccossou ! There are still some tests failing (mainly all the ml related tests using the tableloader), which has to be fixed. I will do it the next days and open an PR then (including the additional unit tests).

ccossou, 2023-11-08:
True that I only ran the test_table_loader. Didn't realize this could be used elsewhere. 

I assume the broken test in the ml related tests are only due to the name change of the traitlets parameters. We shouldn't have any other type of failures as we're suppose to keep the traitlets mecanism intact.

closed 2024-01-04 by maxnoe