cta-observatory/ctapipe #2174: `run_tool` default arguments

nbiederbeck, 2022-12-21:
https://github.com/cta-observatory/ctapipe/blob/a633e0fb7d0620266c77d6fd68cf085314976b42/ctapipe/core/tool.py#L542

We only use `run_tool` in test files:
```
$ rg run_tool -l
ctapipe/conftest.py
ctapipe/core/__init__.py
ctapipe/core/tool.py
ctapipe/instrument/conftest.py
ctapipe/tools/tests/test_process_ml.py
ctapipe/tools/tests/test_process.py
ctapipe/tools/tests/test_merge.py
ctapipe/tools/tests/test_train.py
ctapipe/tools/tests/test_tools.py
ctapipe/tools/tests/test_apply_models.py
ctapipe/core/tests/test_tool.py
ctapipe/core/tests/test_traits.py
```
I assume it is not used externally.

I have two points to discuss:

1. It happens that `cwd` is not set properly and tests clutter the working directory with (prov-)log-files etc. Maybe it should have no default argument, so it gets set correctly by the developers implementing the tests?
2. `raises` is by default `False`, which I think is the wrong choice. One can either check correctly for raised errors when writing the tests, or if `assert run_tool([...]) != 0` is really needed, set `raises=False`. I think this is mainly a quality of life change for developers debugging while implementing tests.

What do you think?

maxnoe, 2022-12-21:
>  Maybe it should have no default argument, so it gets set correctly by the developers implementing the tests?

Or the default should be a tmpdir

maxnoe, 2022-12-21:
> What do you think?

Agreed.

nbiederbeck, 2022-12-21:
> Or the default should be a tmpdir

We are using at least `dl1_tmp_path` and `dl2_tmp_path` so I would say one should set it accordingly to whatever gets tested.

Not sure though if different temporary paths are needed at all

PR #2175 (2022-12-21): Change default behaviour of `run_tool`
Fixes #2174
Fixes #2110
closed 2022-12-21 by maxnoe