cta-observatory/ctapipe #2184: Create changelog entries for all PRs merged since 0.17.0

maxnoe, 2023-01-05:
With the new changelog system in place, we should make the effort of providing entries for all PRs merged.

kosack, 2023-01-05:
Do you suggest trying to export the existing PR titles into towncrier format, or could we just cut-and-paste the ReleaseDrafter version for old PRs, and just use towncrier for the new ones?

maxnoe, 2023-01-05:
I was talking about creating those snippets for each PR with a meaningfull summary, not just the title.

kosack, 2023-01-05:
I guess we could extract the PRs using the github API and convert them.  E.g. using the gh command-line tools.   However, thats a lot of work and many have non-meaningful descriptions, so it might just be easier to create a manual changelog from the existing ReleaseManager for the past few releases, and just use better version *from now on*

Note:
```bash
>  gh pr list --state merged --limit 10000 | wc -l
1204
```
Though many of the early ones are not really PRs, but direct commits from before when we started using PRs...  They are also mostly uncategorized

maxnoe, 2023-01-05:
That's more than I intended. I just meant since the last release 0.17 since we now have a few entries in the changelog but not all, since the changelog PR was merged after 0.17 was released and a number of PRs were merged.

I will open a PR

kosack, 2023-01-05:
Yes, that sounds reasonable to start with v0.17.

StFroese, 2023-01-10:
closed by #2186

closed 2023-01-10 by StFroese