cta-observatory/ctapipe #2212: matplotlib 3.6.3 reveled (or has?) a bug when changing the norm of CameraDisplay after adding a colorbar

nbiederbeck, 2023-01-12:
In https://github.com/cta-observatory/ctapipe/pull/2207/commits/a6dc89238d33f09d75627ec864d34236b6114a57 I introduced a test that is skipped in matplotlib 3.6.3 showing the difference.  This test should run regardless of the version.

I tried to narrow it down, and it seems like the values that should get scaled are the values of the colors inside the colorbar, not the values they represent. E.g. the "colors" are always 0--1, but the values they represent are like 10--100 or whatever.

When adding the colorbar to images with values 10--100, the colorbar-values are 0--1. Changing the norm to logarithmic now tries to scale 0--1, which is of course not possible (`log(0, 1) == -inf, 0`).

I don't know if this is really a bug in matplotlib or in our way of handling the norm on patches of pixels.

maxnoe, 2023-01-25:
Definitely a bug in matplotlib, issue: https://github.com/matplotlib/matplotlib/issues/25077

maxnoe, 2023-01-26:
Fix was merged, to be released with 3.7

maxnoe, 2023-06-05:
matplotlib 3.7 is released since quite a while, let's close this